vitotai/BrewPiLess #1: add Mashing

universam1, 2016-09-14:
First off all thank you for the great work on this, I was in the middle of doing the exact same thing when I discovered your work :smile: 

My main use case is though mashing, not fermenting. I don't see a reason why this should not work, do you?
Though, there is a max temperature limit in code which prevents this usage.
Could this be removed?

On long term, it would be awesome to add a button to make this easier by not supplyling the date/time but a time diff, so for example 70min (which would do current time + this value)

I think this changes would increase the use case so much!

Thank you for considering!
Sam

vitotai, 2016-09-14:
Well,
That is another project.

Check these two:
https://github.com/vitotai/BMESP8266
https://github.com/vitotai/BrewManiac

I've already ported Arduino Part to ESP8266. I already used it in really brew.

universam1, 2016-09-14:
Thanks for reply!

As far as I understand it still does require an Arduino? Which one you're referring to?

vitotai, 2016-09-14:
well,
"I've already ported Arduino Part to ESP8266."

It is not on github.com yet.

universam1, 2016-09-14:
That sounds awesome! Would you be willing to share? Thank you so much!

vitotai, 2016-09-14:
Not yet. There are something to be improved.

closed 2016-11-17 by vitotai