vitotai/BrewPiLess #3: Web interface crach

vstoms, 2016-11-17:
Hi,

Cool project you have here, but sometimes the web interface crash:

`al(function(){if(!a){controllerError()}BWF.send("l");a=false},5000)}window.onclick=function(c){var d=document.getElementsByClassName("dropdown-content");var b;for(b=0;b<d.length;b++){var a=d[b];if(a.classList.contains("show")){a.classList.remove("show")}}};/*]]>*/</script>
<body onload=init()>
<div id="lcd" class="lcddisplay"><span class="lcd-text">
<span class="lcd-line" id="lcd-line-0">Live LCD waiting</span>
<span class="lcd-line" id="lcd-line-1">for update from</span>
<span class="lcd-line" id="lcd-line-2">script...</span>
<span class="lcd-line" id="lcd-line-3"></span></p><p>
</div>
<div class="dropdown">
<div id="myDropdown" class="dropdown-content">
<a href="/control.htm">Temperature Management</a>
<a href="/log">Data Log</a>
<a href="/setup.htm">Device Setup</a>
<a href="/config">System Config</a>
</div>
</body>
</html>`

vitotai, 2016-11-17:
What browser are you using? First of all, BrewPiLess doesn't work with Microsoft Edge and IE. Second, if the browser crashes, please contact the company like Microsoft, Google, and Apple.
If what you mean is that the web interface stops updating data, that is 'normal' because the internet connection breaks from time to time. Just reloading will resolve the issue.
Regularly check and reconnect the link (by program) will alleviate this issue, but I found the WebSocket library was not stable and  easy to crash after several reconnecting attempts. This is the main reason for which I changed back to Server Side Event which is not supported by Edge and IE.
There should be room for improvement. Let me check.

vstoms, 2016-11-17:
Hi,

I'm using Chrome (Also tested from an Android device). If I wait 30min and trying to connect then I get the frontpage (The led display), but if I try to swith to "Temp controll" then I get the html code:

```al(function(){if(!a){controllerError()}BWF.send("l");a=false},5000)}window.onclick=function(c){var d=document.getElementsByClassName("dropdown-content");var b;for(b=0;b<d.length;b++){var a=d[b];if(a.classList.contains("show")){a.classList.remove("show")}}};/*]]>*/</script>
<body onload=init()>
<div id="lcd" class="lcddisplay"><span class="lcd-text">
<span class="lcd-line" id="lcd-line-0">Live LCD waiting</span>
<span class="lcd-line" id="lcd-line-1">for update from</span>
<span class="lcd-line" id="lcd-line-2">script...</span>
<span class="lcd-line" id="lcd-line-3"></span></p><p>
</div>
<div class="dropdown">
<div id="myDropdown" class="dropdown-content">
<a href="/control.htm">Temperature Management</a>
<a href="/log">Data Log</a>
<a href="/setup.htm">Device Setup</a>
<a href="/config">System Config</a>
</div>
</body>
</html>```

vitotai, 2016-11-17:
That's weird. I am using Chrome during the development.
The issue you show is like that the browser had trouble rendering the page because the content is incorrect. It might be an issue of the ESPAsyncWebServer and ESPAsyncTCP. I seem to have similar issue sometimes if I turn on the development console of Chrome (for other project).
Two things I am going to do are
1. add Cache control, so that once the browsers get correct files, it won't need to get them again.
2.Check new version of  ESPAsyncWebServer and ESPAsyncTCP.

vstoms, 2016-11-19:
After the update, things looks much faster and better. I will do some testing and see if I get the same problems again.

Keep up the good work!

vstoms, 2016-11-22:
Update: I think you can close this issue now. I can`t reproduce the errors after the change you made.

closed 2016-11-28 by vstoms