vitotai/BrewPiLess #5: adding support for iSpindel

universam1, 2017-02-23:
I'm the owner of the project https://github.com/universam1/iSpindel that is a electronic Hydrometer which measures the fermentation of the wort. I'm working on creating a fermentation temperature controller called iRelay based on that development that makes use of the data coming from the iSpindel that is floating inside which is in particular

- temperature inside the fluid which is way higher accurate than measured outside with a sensor
- making use of the SG / Plato to adjust the fermentation desired temperature. Therefore you can automatically rise the temp to cope with Diacetyl for instance.

I realize that you've created already a very nice project and I wonder if you would like to collaborate in this regard instead of creating something new. I've received already some requests for this https://github.com/universam1/iSpindel/issues/7#issuecomment-279343482

That would include 

- receiving the incoming json data from the iSpindel as a new temp sensor on a specific TCP port, decoding it and displaying.
- Forwarding this data to an IoT service, kind of proxy.
- further replacing the time based planning with event based on the SG measured. For instance on 80% of fermentation go to temp X.

The above functionality is already coded by me.

Let me know if you are interested.
Sam

vitotai, 2017-02-23:
Hi Sam,
I am happy to do that. In fact, I might do it even without your request.

These two can be done easily and quickly.
 * receiving the incoming json data from the iSpindel as a new temp sensor on a specific TCP port, decoding it and displaying.
 * Forwarding this data to an IoT service, kind of proxy.

I would suggest to use TCP:80, the web interface. iSpindle can already post to an URL. We just need to create a format. It can be done in minutes, literally.

 * further replacing the time based planning with event based on the SG measured. For instance on 80% of fermentation go to temp X.

This will take longer because we need:
 * a new interface that can be used to specify the condition. The condition may be time, gravity, or combined of both.
 * considering the case that gravity report missed for a certain time.

I am not at UI design. Should you have any idea, please shed some light.

universam1, 2017-02-24:
Hi @vitotai 
This sounds really nice! Happy to work together and take your really nice developed project to further usecase! There are many requests for the iRelay functionality pending and I'm fine with shifting to your project!

> I would suggest to use TCP:80, the web interface. iSpindle can already post to an URL. We just need to create a format.

I've implemented a bunch of protocols already. Sure I can write against every port and URI you wanna have. We can start with the HTTP Post to get it flying, but eventually I am in favor of using a plain TCP socket because of runtime costs. Plainly the on-time drops without the HTTP overhead which by the end of the day results in a quite dramatic battery lifetime improvement. 
But okay lets start with HTTP Post. The request will look like this json string:

`{"name": "iSpindel01", "id": "XXXXX-XXXXXX", "temperature": 20.5, "angle": 89.5, "battery": 3.87}`

Currently I'm still working on a clean implementation of the conversion to gravity in firmware, this would extend the json definition.

So our plan could be like this:

1. create a Post URI where to send the json data, parse the request
2. add the iSpindel data as new temperature sensor input 
3. use the inbuilt DS18b20 as sanity checks if the iSpindel is not accessible (limit the diff to x°K)
4. work as a relay  / proxy - forward the data to Ubidots or another storage
5. extend the UI to take gravity as optional condition

Your comments!

How do you want to split the work up?

vitotai, 2017-02-24:
If overhead of HTTP is concerned, maybe UDP would be better than TCP. By using broadcast UDP, no setup will be necessary of iSpidel. By using a make-up and predefined HTTP header instead of using HTTP client libary, I think it should take less overhead. But, let's start with HTTP for easy and simple debugging.

I have a few questions:
 1. Is it possible to calculate gravity in iSpindel? The value of angel is not meaningful to user. I would bet that users prefer to know the gravity instead of angel. If the gravity is not calculated by iSpindel for power saving, it must be calculated somewhere else and the parameters to calculate should be provided.
 2. What is the "id": "XXXXX-XXXXXX" for? Are we going to support multiple iSpindle? 

>  1.create a Post URI where to send the json data, parse the request

This is easy, but "using them" is more complicated. What are we going to use them?

> 2.add the iSpindel data as new temperature sensor input

It's easy if we just want to display it. I don't think it's a good idea to use the senor as an "input" to temperature management.

> 3.use the inbuilt DS18b20 as sanity checks if the iSpindel is not accessible (limit the diff to x°K)

Easy or not. A new UI? What to do if it happend?

> 4.work as a relay / proxy - forward the data to Ubidots or another storage

The code and and the structure is already there. Just adding some more lines of code, it will be done.

> 5.extend the UI to take gravity as optional condition

This will be a complicated one. Should we start from User cases? 
Currently, the way to specify the temperature schedule in "Beer Profile" mode is by specifiying a list of *"**N**'th day at temperature **T**"*. Given a time **n**, the temperature can be derived by finding **N1** and **N2** where **N1** <= **n** < **N2** and the desired temperature will be **T1+(T2-T1)/(N2-N1)x(n-N1)**. IMO, it's a simple and elegant way. The best part of it is that user can specify the speed of temperature change while the program handles it in the same way. However, I don't think of a way to "extend" it.

universam1, 2017-03-01:
Sorry for late answer, I was brewing and setting up the iRelay to demonstrate like this: http://hobbybrauer.de/forum/viewtopic.php?p=195277#p195277

> Is it possible to calculate gravity in iSpindel?

Yes, I'm workin on it, ETA not too far.

> What is the "id": "XXXXX-XXXXXX" for? Are we going to support multiple iSpindle?

Not necessarily, but to display and forward to Ubidots

> 1.create a Post URI where to send the json data, parse the request
This is easy, but "using them" is more complicated. What are we going to use them?

Just referring to the backend to connect from the iSpindel.

> 2.add the iSpindel data as new temperature sensor input
It's easy if we just want to display it. I don't think it's a good idea to use the senor as an "input" to temperature management.

That's _the_ critical important point we have to agree on. Thats why I set up and please take a look at the very live running fermentation control of an iSpindel sending the temp to the iRelay which controls based on it. It is outside so I'm heating instead of cooling, but this doesnt matter.
https://app.ubidots.com/ubi/public/getdashboard/page/Q25QJuWDcPmc-Nh_lnMff_ed9V8

- blue is the ambient temperature, 
- red the wort messured by iSpindel and 
- orange the temp of the kettle outside

Lesson I learned:

- Here is the point, the kettle outside temp is really far away from the _real_ wort temperature, and here is the big benefit of the iSpindel. For instance the yeast creates heat and the iSpindel floats just in the very center of yeast activity. Perfect to control better off-flavors.
- It is perfectly possible to control the temp based on the data sent from the iSpindel

BTW. this is altogether 2 weeks of runtime, now the 4. day of real brew.

> 3.use the inbuilt DS18b20 as sanity checks if the iSpindel is not accessible (limit the diff to x°K)
Easy or not. A new UI? What to do if it happens?

` if (desiredT > sensorT && kettleT < desiredT + HEATTOLERANCE)  { heat() }`
Thats all I do for now and the result you can see

> Given a time n, the temperature can be derived by finding N1 and N2 where N1 <= n < N2 and the desired temperature will be T1+(T2-T1)/(N2-N1)x(n-N1). IMO, it's a simple and elegant way. 

Fully agree, this is elegant. We might think of optional substitue Nx with gravity, then your logic could stay unchanged. But this is thoughts for future ideas, I dont think this is critical for now.

vitotai, 2017-03-02:
How often does iSpindle report temperature? 

The core of BrewPiLess is BrewPi, which runs the temperature control algorithm. I don't think the temperature reported form iSpindle serves a good source for BrewPi's algorithm which requires more frequent updating of temperature readings, like very few seconds.

It seems to me that you are asking for a new way of temperature control based on the iSpindle sensor. I see that you have a good temperature control program. Then, people have a choice other than BrewPi(Less) if they want. IMO, the temperature control algorithm of BrewPi is the most valuable part of BrewPiLess. Removing that part makes BrewPiLess Useless.

universam1, 2017-03-02:
The interval is configurable, but of course not every second. Nevertheless this is not needed at all for this kind of super slow PID of controlling required here. 

Did you analyze the demonstration I created for you?

As you see `Tau` is at 10min, 600s. For a proper PID control you need `Tau >= 10x Ta`. This case we have `Ta` of 15s so we are at 40x, which is more than sufficient.

No I dont think it makes it useless at all. It just means using a different source of truth, a pretty trivial impact.

vitotai, 2017-03-02:
How about the battery life for this setting?
I thought it reported every hour.

That seems to doable, but I think the PID parameters should be tuned for that. Considering the situation of falling back to connected sensor, some options for better control,  and hacking of BrewPi algorithm ( I just traced the code I need to change.), it needs more time than gravity-based temperature schedule. I would put it on the list after gravity based schedule.

BTW, I already created a "preview" branch which logs gravity and "auxiliary" temperature. The only thing missing is the input of iSpindle. Should we finalize the format of reporting format?

universam1, 2017-03-07:
> How about the battery life for this setting?

With currently insane interval of 15s I have a runtime of about 14days. I'm convinced we can go up to 1min or more without a doubt, the battery life scales pretty exactly reziprok with the interval freq. Some testing to be done.

> it needs more time than gravity-based temperature schedule. I would put it on the list after gravity based schedule.

I'm okay to take care of it.

> Should we finalize the format of reporting format?

I'm focusing hard to provide the gravity calculation from firmware in recent time. Yes we can finalize the format:

- HTTP Post
- URL: up to you
- format: json: 
`{"name": "iSpindel01", "id": "XXXXX-XXXXXX", "temperature": 20.5, "angle": 89.5, "gravityP": 13.6, "battery": 3.87}`
- port: 80

Important point, regarding to the proxy functionality, the BPless has to open an AP where the iSpindel can connect. Is that available already?

vitotai, 2017-03-07:
I am using "/gravity" as URI. One problem here is that users have to figure out the IP of BrewPiLess. BrewPiLess does support mDNS, but involving mDNS might increase power consumption of iSpindle. (Maybe we should consider broadcast UDP seriously.)

The unit of temperature is always Celsius, right? 

BrewPiLess can be configured to update to IOT services. I already added a few fields to the report. 
| Specifier   | output  |
| -------------- |:-------------|
| %b         | Beer temperature   |
| %B         | Beer setting   |
| %f         | fridge temperature   |
| %F         | fridge setting   |
| %r         | room temperature   |
| %g         | gravity   |
| %a         | auxiliary temperature   |
| %v         | external device voltage   |
| %u         | UNIX timestamp of last gravity update   |

Please check the detail here:
https://github.com/vitotai/BrewPiLess/tree/preview#remote-temperature-logging

universam1, 2017-03-08:
I solved it this way that your controller opens an AP `WiFi.mode(WIFI_AP_STA);` with a specific SSID pattern `String ssid = String("iRelay-") + String(ESP.getChipId());
  WiFi.softAP(ssid.c_str(), IRELAYPASS)` and a commonly known pass, so it can be discovered by the iSpindel. It also serves the proxy forwarding / wifi boost range extension idea.

Okay so you want to introduce a new protocol.

vitotai, 2017-03-08:
Using softAP seems to be a very good idea. Let me test how it works.

Another question: should we use some kind of low pass filter on the gravity reading? I think the reading might be affected by unexpected movement of fermenters. It's OK if the data is displayed and read by human. It might be an issue if the gravity reading is used to decide the fermentation temperature setting.

universam1, 2017-03-08:
Good point, makes sense. I think you could really average over an hour or even more, from the practical point of view.
What unit for gravity are you expecting? I'm used to Plato.

vitotai, 2017-03-09:
Specific gravity, like 1.012.
There are formula to convert between SG and plato. However, I don't know if it is valid to convert SG values smaller than 1. I learned that some mead can have FG smaller than 1.

vitotai, 2017-03-11:
Is it possible for you to give me some gravity data? I would like to check the effectivity of the filter.

vitotai, 2017-03-13:
A problem of moving average is memory usage. 
Can we use single pole IIR low-pass filter?  It is simple as
y[n]=y[n−1]+b(x[n]−y[n−1])

We need to decide the value of "b".  How do you think?

I almost finish the gravity based schedule. Do you have iSpindel code ready for me to test?

universam1, 2017-03-14:
The data from the iSpindel is very stable and almost no noise, I wonder if a filtering is needed at all. Probably a filtering over an hour would be sufficient and a running average could be sufficient and takes no memory:

```
float alpha = 0.1; // factor to tune
value = alpha * measurement + (1-alpha) * value;
```

Will have a sample code in a short time.

vitotai, 2017-03-14:
That's what I am thinking. 
Are you going to report every minute or make it a configurable option? From my point of view, one minute period is overkill if the temperature is not used in the control algorithm. If the report period can be changed, maybe I need to make the value "alpha" configurable.

I've finished the gravity based schedule and now I am conducting stability test. My ESP8266 is not stable in STA_AP mode. I am not sure what went wrong and have decided to upgrade ESP8266/Arduino environment and ESPAsyncWebServer libraries. I am testing it under STATION mode only now for cross verification. (BTW, I found that closing HTTP connection without calling "end()" of HttpClient can crash ESPAsyncWebServer quickly.)

universam1, 2017-03-15:
end() is not always sufficient since it leaves the TCP connection orphaned. There are some MR regarding this open, I tested it and it helps. Indeed I can confirm that sometimes it crashes still. Thats where I bother currently with. It must be possible since the meshing works the same way.

The temperature control over the input from the iSpindel is not optional for me, once you are running the setup you will understand why. But the interval is adjustable of course. It has to be evaluated where the sweet spot is.

vitotai, 2017-03-15:
What is MR?  A delay of 100ms before turning off WiFi works for me.
At this point, I am open to using TCP directly as you previously proposed if that simple TCP is more robust. BrewPiLess servers as a web server, so the web interface is already there to be used. One benefit of using web interface is no extra TCP connection necessary. BrewPiLess uses ESPAsyncWebServer/ESPAsyncTCP to enable concurrent multiple connections. The maximum number of connections ESPAsyncTCP supports is 5. Memory resource is precious for ESP8266, and I think 5 is reasonable.

To reduce the memory usage and MIPS requirement to process the logs, a simple binary log format is used. Sparsely changed data is recorded in a different way from that of continuously changing data. If the data is logged every minute, it should be treated as continuously changing data. I have to figure out a way of balance among simplicity, efficiency, and functionality.

vitotai, 2017-03-16:
Struggling on AP+Station mode.
The station lost connection and reconnects every 5 minutes.(logs from serial.)
The lost of connection disappears when a browser establishes a connection to it.
The connection from the browser might fail if it happens at the time ESP8266 reconnecting.

It works fine if SoftAP is not created and the ESP8266 is in station only mode.

Do you have any idea?

vitotai, 2017-04-15:
1. BrewPiLess can run STA+SoftAP mode. Under that mode, it connects to an AP and creates a wireless network at the same time. BrewPiLess works as usual, and no function is missed.

2. My current design is creating an option to enable AP mode. If AP mode is enabled, it will be in STA + SoftAP mode. No mater iSpindel connects to the same AP as BrewPiLess or to BrewPiLess directly, the gravity update request can be processed.

3. My current issues seems to be an issue of my router.

allthepies, 2017-04-15:
So I can operate BrewPiLess in station mode and use NTP, run beer profile + get browser access via my local Wifi and also get iSpindel to connect to the BrewPiLess IP also via the same home Wifi network ?

vitotai, 2017-04-15:
Yes. I googled and found that might be an issue of my router.
It's not a deal breaker.  Everything works, but sometimes I just have to wait for a few seconds or retry for several times to connect to BrewPiLess. Once the connection is established, there is no issue at all.
Therefore, I decide to make it an option. People who have good routers might have luck to run without any issues.

allthepies, 2017-04-25:
Hi Guys,  Any update on the interface between iSpindel and BrewPiLess ?  Specifically on how the gravity can be calculated from the tilt angle output by iSpindel ? Or is the intention for iSpindel to perform the gravity calculation and output it in Plato and/or degrees in it's JSON payload ?

vitotai, 2017-04-25:
I just finish the calibration and plan to give it a try.
You can check the calibration document to find the coefficients. The plato is just calculated by
a0 + a1 * x + a2  * x^2   
where x = tilt
I use a third order formula, and an additional coefficient is derived and an additional term a3 * x3 is added.
It's easy to add a parameter to the genericHTTP. However, I found that the TCP connection is not closed correctly, which results in inaccessibility of the web server of BrewPiLess. I added a  a delay(50) before going to deep sleep.
One issue I have is that the gravity reading drifts a little bit after I pull the sled out when trying pull off the USB connection. Maybe a correction parameter is necessary.

allthepies, 2017-04-25:
Great!  Are you calculating the specific gravity in degrees too (I don't use Plato).

vitotai, 2017-04-27:
I don't use plato, neither.
The conversion is easy by this formula:
sg=1 + (plato / (258.6 -((plato/258.2)*227.1)));

h44r3k, 2017-05-02:
I´m trying to connect my ispindel (v5.12) to bpl (v.2) but can´t get it to work.. I´ve setup the BPL with softAP and the ispindel seems to connect, but i can´t get any data reading.. Last service type i tried was TCP with hostname brewpiless and port 80. Could you please give the steps to make it work?

allthepies, 2017-05-02:
You need a http based service type, can't remember which one off the top of my head.  Tried the CraftBeerPi or Univeral HTTP types ?  Basically you need one where can can supply the server IP, port and URL (which needs to be /gravity ).

h44r3k, 2017-05-02:
I´ve tried both HTTP and CBPi, i can´t get them to work.. What port did you use?

allthepies, 2017-05-02:
IP address = BrewPiLess's IP, Port = 80, URL = /gravity

You were setting the URL to port 80 ?   If you can't get it to work then I'd hook up the spindel to a PC and use the Arduino IDE serial monitor or similar to see what the spindel is complaining about.

allthepies, 2017-05-02:
Oh, and enable iSpindel support in BrewPiLess on the gravity settings page.

h44r3k, 2017-05-02:
The Arduino IDE said version 5.12 of iSpindle had problems with the wifi credentials so i flashed the esp with version 4.1.6 and now the iSpindle connects to the network, but i still can´t read it in BPL. I´ve enabled the iSpindel support in BPL. Now the BPL seems to be crashing after 1-2min

FW 4.1.6
1.5.4(baaeaebb)
Worker run!

Double Reset detected
mounting FS... mounted!
reading config file
opened config file

parsed json
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":30,"Server":"192.168.1.127","API":3,"Port":80,"URL":"/gravity","Vfact":191.80,"SSID":"My_SSID","PSK":"My_Pass"}
going to Config Mode
*WM: Scan done
*WM: DUP AP: My_SSID
*WM: Adding parameter
*WM: name
*WM: Adding parameter
*WM: sleep
*WM: Adding parameter
*WM: vfact
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: 
*WM: Adding parameter
*WM: selAPI
*WM: Adding parameter
*WM: token
*WM: Adding parameter
*WM: server
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: url
*WM: SET AP
*WM: 
*WM: Configuring access point... 
*WM: iSpindel
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started

allthepies, 2017-05-02:
OK, the above log shows the iSpindel in config mode i.e. it's created a local Wifi hotspot and is waiting for you to configure it.  I had lots of problems with the iSpindel in this area!  Try connecting to the iSpindel hotspot and configuring the Wifi/connection properties again.

h44r3k, 2017-05-02:
Hmm, i´ve got it to connect now have and the ispindel is updating every 30sec but after the second intervall, it seems like the BPL crashes.. What BPL version are you running?

{"Name":"iSpindel01","Token":"","Sleep":30,"Server":"192.168.1.127","API":3,"Port":80,"URL":"/gravity","Vfact":191.80,"SSID":"xxxx","PSK":"xxxx"}
woken from deepsleep, normal mode
Spl 0: 86.12
Spl 1: 86.17
Spl 2: 86.13
Spl 3: 86.20
Spl 4: 86.17
Spl 5: 86.17
Spl 6: 86.15

a: 1120	16394	148	absTilt: 86.16	T: 26.17	V: 5.34	owT: 23.81
After waiting 0.00 s, result 3
192.168.1.140

calling DTHTTP 

ERROR genericHTTP: couldnt connect

Final-sleep: 30s; RT:10933

allthepies, 2017-05-02:
I can't remember which version of BPL I tested iSpindel with, I had it running on a testbed only.  I'll give it another go with the latest and greatest BPL version from github.

allthepies, 2017-05-02:
OK, just gave it a go with the latest BPL codebase and the iSpindel can repeatedly connect and post updates without issue.

Here's the iSpindel output for a couple of successive readings/postings.

```
FW 4.1.7
1.5.3(aec24ac9)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
parsed config:
{"Name":"iSpindel000","Token":"","Sleep":20,"Server":"192.168.1.101","API":3,"Port":80,"URL":"/gravity","Vfact":191.80,"SSID":"BTHub5-MZ7P","PSK":"e3c3c8673a"}
woken from deepsleep, normal mode

a: 0	0	0	absTilt: 25.60	T: 22.50	V: 4.70	owT: 25.40
After waiting 2.70 s, result 3
192.168.1.184

calling DTHTTP 
genericHTTP: posting
POST /gravity HTTP/1.1
Host: 192.168.1.101
User-Agent: iSpindel000
Connection: close
Content-Type: application/json
Content-Length: 86

{"name":"iSpindel000","ID":"2024627","angle":25.60,"temperature":25.40,"battery":4.70}HTTP/1.1 200 OK
Content-Length: 2
Content-Type: application/json
Connection: close
Accept-Ranges: none

{}
Final-sleep: 20s; 
```

vitotai, 2017-05-02:
Don't rush!
You will get into trouble without modification the iSindel code.
I will post the solution/procedure later.

h44r3k, 2017-05-02:
Ok, i´ll wait 👍 Thanks for trying to help me allthepies!

vitotai, 2017-05-03:
I've updated the code.
However, I will be able to test it after two or more weeks.

Please read the iSpindel support section carefully. (I did manage to build an image of iSpindel and put it in /extra folder, but I worked around  to solve a linking issue.)

allthepies, 2017-05-03:
Thanks Vito.  My setup worked because I'd spotted the missing delay in the iSpindel code and had added one in before the spindel went to sleep.   That explains that then :)

I'll load up the latest BPL version this evening and give it a go.

h44r3k, 2017-05-03:
Thanks! Seems to be working just fine now 👍 I´m looking forward to test BPL and iSpindel with my next beer!

allthepies, 2017-05-03:
<off topic>

h44r3k: From the iSpindel trace you posted above I see that it was using SDK v1.5.4.   How did you setup your dev environment to get that level of SDK ?    Or were you using a prebuilt .bin file from the iSpindel project. ?

h44r3k, 2017-05-03:
I used the prebuilt .bin because i couldn´t get the ispindle to compile from Arduino IDE

allthepies, 2017-05-03:
Ah, OK.   I build it direct from the Arduino IDE, can't remember having a problem setting it up.  I'll look into getting a dev environment using a more recent SDK.

allthepies, 2017-05-04:
I gave the latest BPL a go last night (with the enhanced iSpindel support).  Looks good, just a couple of early comments:

1. When entering the gravity settings page for the first time then an error alert popup appears with a 404 error.  I guess it's attempting to fetch settings which don't exist first time (was fine after settings saved).  Perhaps this initial state could be handled more gracefully.

2. The Gravity scale on the chart doesn't seem to match the reported gravity.

vitotai, 2017-05-07:
1. The 404 alert should be removed. (I thought I had removed it.)
2. Please provide more information. The chart is updated once per minute. The reported value next to the buttons is updated when iSpindel reports. They might disagree to each other, but it should be minor in real application.

lekrom, 2017-06-08:
Dear @vitotai, @universam1,

Big fan of both of your work. Great that you are working together on this one...
I have recently started homebwrewing (my first batch is fermenting now). I have built my own single vessel system using a brewmaniac / bmesp8266 controller (which works pretty sweet). I have also built an iSpindel. Fortunately it is winter here in South Africa, and the temperature is spot on for fermenting ales at the moment. This will not be the case come summertime. Therefore, I am planning to build a fermentation chamber using a Brewpiless next. One question, is there any plans to enable using the iSpindel temperature measurement as the "Beer temp" measurement for Brewpiless? Would this be practical? I assume you would likely need to set the iSpindel to report a bit more frequently which would impact battery life.

zandoga, 2017-07-02:
Excellent so far with Brewpiless and the iSpindle, however I've been trying for a few days to pair them together with no success. It seems that iSpindel connects to BrewPIless however the SG is not shown in the BrewPiLess webpage.
I appreciate your feedback. Thank you.

-IDE 1.8.2
-BrewPiLess v2.0
-iSpindel 5.1.2
-followed iSpindel Support instructions from BrewPiLess readme, including the addition of delay(50) in iSpindel.ino
-both are connected to the same router

FW 5.1.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"dRKBpgNNdKVK8mt1R49BUXMdNpW2Hm","Sleep":150,"Server":"192.168.1.100","API":3,"Port":80,"URL":"/gravity","Vfact":200,"SSID":"XXXXXXXX","PSK":"XXXXXXX","POLY":"0.01636441875*tilt^2-1.2118026*tilt+23.47368037","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 50.91
Spl 1: 50.72
Spl 2: 51.00
Spl 3: 51.02
Spl 4: 50.83
Spl 5: 50.93
Spl 6: 51.08

a: 822	12514	10100	absTilt: 50.95	T: 29.90	V: 4.07	owT: 30.06	Gravity: 4.21
After waiting 0.00 s, result 3
192.168.1.101

calling DTHTTP 
genericHTTP: posting
POST /gravity HTTP/1.1
Host: 192.168.1.100
User-Agent: iSpindel000
Connection: close
Content-Type: application/json
Content-Length: 112

{"name":"iSpindel000","ID":"16729250","angle":50.946877,"temperature":30.0625,"battery":4.065,"gravity":4.21135}
Final-sleep: 150s; RT:4967

I also tried to connect to BrewPiLess as AP most of the time seems to connect but not always, but the SG does not show up un the BrewPiLess webpage:

FW 5.1.2
2.0.0(656edbf)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file

parsed json
offsets not available
parsed config:
{"Name":"iSpindel000","Token":"dRKBpgNNdKVK8mt1R49BUXMdNpW2Hm","Sleep":150,"Server":"192.168.4.1","API":3,"Port":80,"URL":"/gravity","Vfact":200,"SSID":"brewpiless","PSK":"brewpiless","POLY":"0.01636441875*tilt^2-1.2118026*tilt+23.47368037","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 72.24
Spl 1: 72.07
Spl 2: 71.96
Spl 3: 71.74
Spl 4: 71.61
Spl 5: 71.82
Spl 6: 71.37

a: 3724	15162	4028	absTilt: 71.84	T: 32.37	V: 4.06	owT: 31.50	Gravity: 20.88
After waiting 0.00 s, result 3
192.168.4.2

calling DTHTTP 
genericHTTP: posting
POST /gravity HTTP/1.1
Host: 192.168.4.1
User-Agent: iSpindel000
Connection: close
Content-Type: application/json
Content-Length: 110

{"name":"iSpindel000","ID":"16729250","angle":71.842896,"temperature":31.5,"battery":4.06,"gravity":20.877611}
Final-sleep: 150s; RT:4881

vitotai, 2017-07-02:
I tried your data by using "curl" command to simulate iSpindel.
The data was shown, but wrong. 

[image]
[image]

zandoga, 2017-07-02:
Your Gravity Sensor interface looks different. I am running v2.0. How do you run the "curl" command? Any idea why the data is not shown at all?

[image]

zandoga, 2017-07-02:
I run the curl and it seems that it's working fine, but not from the iSpindel. I'll keep digging...

[image]

vitotai, 2017-07-03:
You might try a IP sniffer, like wireshark.
Or, you might need to turn on debug message output by recompile BrewPiLess. the debug message output option is in espconyfig.h

BTW,
I just pushed updated code.

allthepies, 2017-07-03:
Vito,  can you summarise the changes in the latest update ?

universam1, 2017-07-03:
nice job done @vitotai !

With the new Firmware since 5.x the iSpindel calculates gravity by itself, so you can drop that conversion from your firmware.

Greetings!

vitotai, 2017-07-03:
hi universam1,

You have done a great job. The using of tinyexpr is a brilliant idea.
There is an option to using gravity data from iSpindel, so it's ok to keep the code there.

Regards

vitotai, 2017-07-03:
allthepies,

check here: http://vito.tw/?p=720
the detail is in the readme@Github.

allthepies, 2017-07-03:
Thanks Vito.

I'm running it on my test rig, will see how it goes before upgrading the "prod" BPL ;)

zandoga, 2017-07-04:
Hi Vito. It's working now and I am not sure what was the issue. I was using a laptop with debian, which crashed. Reinstalled everything from scratch on both iSpindel and BrewPiLess uising a windows machine and it's all good now. I like your new updates. Brewing party tomorrow and I can't wait to put these toys to work. You have a happy 4-th!

vitotai, 2017-07-04:
Hi zandoga,
Some reported that he couldn't make it work until he installed a new installation of Arduino IDE. It might has something to do with the libraries.
I am hopping to migrate to PlatformIO. Arduino IDE starts to make me sick.

[17 more replies]