vitotai/BrewPiLess #7: State after a power outage

allthepies, 2017-04-09:
If the power is cut while the BrewPiLess is running in Beer Profile mode then upon power resumption (and subsequent Wifi reconnection) the s/w goes into "Off" mode.  I thought the current operating mode was supposed to resume after a power outage ?

vitotai, 2017-04-09:
The state information is stored in EEPROM by BrewPi core.
The setting should not changed in power recycle. Mine won't.
The data in EEPROM might be corrupted so that the mode was reset.

allthepies, 2017-04-10:
Thanks Vito, I shall experiment further.

vitotai, 2017-04-11:
Thanks.
A few words
1. Formatted JS is good for human but bad for machine. Compressing the JS source not only saves a few kilo bytes but also makes downloading faster.
2. Regarding the Date Format, I've made some changes, and you can get it from "preview" branch. Basically, it's simple for the date/time in the profile table. However, it is a little complicated for chart axis. JS doesn't have a SHORT localized date format function, and I think using notation like mm/dd or dd/mm is better than default "Apr 4".  My solution is kinda dirty but won't work under certain condition: I remove the "year part", like 2017. It will not work for people who don't use Common Era, like Japanese.
3. Again. Please check "preview" branch, I've fixed some bugs and issues. They are not critical, but bugs are bugs.
4.Regarding the EEPROM issue, I have thought of re-implement it like Thorrax does. (The BrewPi core was ported by Thorrax.) He uses SPIFFS to store EEPROM data instead of the special section of flash. It solves wear-leveling issue but doesn't improve robustness. IMHO, it degrades robustness by introducing SPIFFS. I have not dig into this part, but I think there might be some bugs or design issues about the EEPROM, because I usually need to "Erase" setting before setting it.

closed 2017-06-19 by allthepies