vitotai/BrewPiLess #8: Upgrade to v2.0 issues

allthepies, 2017-04-17:
Hi Vito,

I've just upgraded to V2.0 and am having lots of problems.

1. The "LCD" display top left of the main page remains in "Live LCD waiting for update from script..." status and doesn't change.
2. If I try and perform any operation which requires authentication then the default brewpi/brewpi user/pass is rejected and the authentication dialog just pops up again.  Tried multiple browsers, same result!

Help !

vitotai, 2017-04-17:
1. try reload again.
Please open the Javascript console and get the log for me.
for Chrome
COMMAND + OPTION + "J" on Mac
CTRL+SHIFT+"J"  on Windows.
 
2. The default username and password changed to brewpiless/brewpiless
However, if you have ever changed the settings by using previous version, the user/pass won't changed.

allthepies, 2017-04-18:
I will have another go this evening Vito.  I did check the Chrome JS console and there was an error, but related to not finding a saved beer profile so didn't think it was significant.

Regarding the changed userid/password, looks like you need to update the README file :)  It still specifies the "old" credentials.

vitotai, 2017-04-18:
I tried to reproduce by cleaning the flash of my D1 mini. It seems to work just fine but one catcha:
The time it spend to switch from STA to STA_AP mode seems to be longer than it used to be. I am not sure if it has something to do with new ArduinoESP8266 framework.

Please make sure you force the browser to clean the cache, and if it still doesn't work. Erase the flash.

allthepies, 2017-04-18:
Did a Chrome cached files delete and it's working :)

closed 2017-06-19 by allthepies