vitotai/BrewPiLess #11: Manually remove iSpindel readings

j0hnsmith, 2017-05-30:
Some invalid iSpindel data was picked up by BPL while I was configuring an iSpindel, is there any way I can manually remove these readings (clearly SG of 16.09 is not valid)?

[image]

gmoyle, 2017-05-30:
Greg Moyle liked your email
Spark by Readdle

vitotai, 2017-05-31:
You can "click" the number after "SG:" to update the SG reading. 
However, the data was written into the log, and  there is no way to remove it.

I am not sure the correct range of SG reading, so I didn't add checking of SG reading. Is 0.8 ~ 2.0 a reasonable range?

j0hnsmith, 2017-05-31:
SG of 1.135 is a potential ABV of 18.4% so I think 0.9 - 1.2 is a reasonable range, I can't forsee any circumstances where a brewer or winemaker would go outside that range.

vitotai, 2017-05-31:
filtering out invalid is simple and easy. However, I bet there will be users who claim no connections between BrewPiLess and iSpindel when they are "testing".