vitotai/BrewPiLess #12: iSpindel (Aux) Temp not displayed correctly

lekrom, 2017-06-23:
Dear @vitotai 

I stumbled upon a small bug. The iSpindel (Aux) temperature is not correctly displayed on my BPL (chart or legend). I setup BPL (in STA-mode) with static IP and setup iSpindel to match. I also setup remote logging (of all available variables) to ubidots. The aux temperature is always displayed as 17.53 (does not change) in BPL. Funny enough, its correctly displayed on my ubidots dashboard. Therefore, I expect it is bug in the display, not the comms.So far, everything else seems to display fine, I have not hooked up the DS18B20's yet (still in the mail).

Regards,
Lekrom

vitotai, 2017-06-23:
Hi,
Maybe something goes wrong, I will check this issue later.
Mine is like this:
[image]

It does change in a weird way.

lekrom, 2017-06-23:
Ok. Thanks for the reply. Great work BTW...

vitotai, 2017-07-02:
I found the bug. It is a stupid but common mistake by using "&&" when it should be "&".
I will fix it shortly.

lekrom, 2017-07-03:
Great, Thanks, I have made a few of those types of mistakes... I will try it out tonight.

lekrom, 2017-07-03:
Thanks @vitotai , appears to be working correctly.

closed 2017-07-03 by lekrom