vitotai/BrewPiLess #13: Conflicting Libs

lalo-uy, 2017-06-29:
I loved this project. Tried the precompiled bins but want to compile myself.

There is a collision between  ESPAsyncWebServer.h  and ESP8266WebServer.h included by WiFiManager.h.
Just post one of the redefinition.
WiFiManager is from  https://github.com/vitotai/WiFiManager

the other libs are from the zip that came with the src.
Arduino IDE 1.6.9   with ESP 2.3.0

Any clue?

Thanks in advance

Ed

```
In file included from /home/lalo/Arduino/libraries/WiFiManager/WiFiManager.h:17:0,
                 from /home/lalo/Arduino/BrewPiLess/BrewPiLess.ino:17:
/home/lalo/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WebServer/src/ESP8266WebServer.h:30:19: error: redeclaration of 'HTTP_ANY'
 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
                   ^
In file included from /home/lalo/Arduino/BrewPiLess/BrewPiLess.ino:7:0:
/home/lalo/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:55:3: note: previous declaration 'WebRequestMethod HTTP_ANY'
   HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_DELETE, HTTP_PUT, HTTP_PATCH, HTTP_HEAD, HTTP_OPTIONS
```

vitotai, 2017-06-30:
I don't know how you get this error if you hadn't modified the code.

At BrewPiLess.ino:17, it is
#include "Menu.h"

WiFiManager use built-in or default synchronous web server, ESP8266WebServer, but BrewPiLes needs the power of asynchronous web server, ESPAsyncWebServer.
Unfortunately, both libraries define the same http method because they do the same jobs.
To solve this, I have to put the code related to WiFiManager to another file, WiFiSetup.cpp.

BTW,
use the libraries in libs/libs.zip
I have modified some of them.

lalo-uy, 2017-06-30:
Well after several Libs update and clearing some duplicated ( xxxx and xxxx-master ) compiles ok now.

My ferm have a cooler inside with cold water from a cold bank.
So I want to do 3 changes
1) no fridge on delay , since it controls an electro valve not a compresor.
2) Change Aux temp plot for Door input and use it to show when the compresor of the cold bank is running.
3) Make Door input field available in the remote logs.
 
Many Thanks for a great work.

closed 2017-06-30 by lalo-uy