vitotai/BrewPiLess #14: Add Config.h toggle to support thorrak's PCB

rflemming, 2017-07-07:
Manually changing the config as describe here is easy enough:

http://www.homebrewtalk.com/showthread.php?p=7983783#post7983783

but since there may be quite a few people out there using that PCB it would be nice to simply be able to toggle that in a single place in Config.h and have all the pins defined accordingly.

vitotai, 2017-07-08:
Maybe there are only very few people who need this, like one.
In fact, the configuration was defined, and even the image was built.
It's tedious to maintain too many configuration, I just dropped that.
Nobody ever notices it. Well, adding that takes less time than typing this response and might save some time answer question on the forum like that. Therefore, I will add it back in next version.
Heads up: the next release will migrate to PlatformIO.

rflemming, 2017-07-08:
Well, as the second person who could use this and would appreciate it, I say thank you :)

closed 2017-07-08 by rflemming