vitotai/BrewPiLess #17: Negative Values at brewchart

jalexandre0, 2017-07-15:
Congrats for amazing work. 

Brewchart goes crazy when beer or freezer setpoing goes nevative. I have a chest freezer and do my cold crash at -2ºC or -4ºC. Depends on ABV. 

I tried to look the issue by myself, but my skills with js/html are very limited at time, and I´m not a programmer. More like a curious guy who handle basic code.  
[image]

Can you put me on right direction or help me to look at this? 

Thank you.

vitotai, 2017-07-16:
This is a real issue.
I thought that the fermentation temperature would never be lower than zero, Celsius or Fahrenheit. 
Therefore, the code cannot handle negative value. The encoding and decoding of temperature data must be revised to support negative data.

zandoga, 2017-07-16:
jalexandre0 just use degrees F for now... You won't go negative with F  :)

jalexandre0, 2017-07-16:
@vitotai  

You are right. Fermentation never should drop about zero, but when cold crashing, you can go the lower you can get without freeze the beer.  

I´ll try to play with it, let me know if you want help or more info.

vitotai, 2017-07-17:
To save processing power and memory, the data is encoded to use minimum space. 15 bits, to be exact.
Encoding negative values might be possible by adding encoding and decoding of negative data.
One problem remains: what is the maximum and minimum value? 

BTW, you don't get frozen beer at -4C, do you? I though the beer might be frozen under zero.

jalexandre0, 2017-07-17:
@vitotai 

Is possible to make  Min/Max values dynamic? 

Don´t see a normal setup (fridge, freezer, glycol bank, you name it..)  overshooting min/max inside normal operation, so I guess the tempSetMin and tempSetMax are good enough to charts. 

If not, I simple explaination about how to change hardcoded values are good enough for most of users, I guess. 

You though is not all wrong. I´m not the average brewer.
 
I like to play a lot with Freeze Destilation and high ABV% beers , so I need to freeze  the water on one or other beer to make a high ABV  even HIGHER!!! \;;/ 

But, for normal batchs in normal conditions, you can assume -1ºC (or 0ºC as safer option)  a good to go option for almost everything with some degree of alcohol.

vitotai, 2017-07-17:
It is of course possible, but not practical. 
What if users set maximum to 120C and minim to -110C? 
To handle such extreme, and sometimes meaningless, situations, the encoding and decoding will be more complicated, and thus buggy.

I am going to do something like -99 ~ 225.  Do you think that's enough?

jalexandre0, 2017-07-18:
Sounds goog enough. :)

jalexandre0, 2017-08-11:
Sorry for delay, longer work weeks.

I downloaded and tested new commit, negative values are ok. Thank you for fixing that. 

I guess you can close this issue.

closed 2017-08-13 by vitotai