vitotai/BrewPiLess #18: Extract frontend-code into separate files

tommueller, 2017-07-25:
Hello,

thank you for the great job on this project! I just installed it on my ESP8266 and it works smoothly.

I would like to contribute to the project. I come from the web-frontend world and would like to help to modernize the looks of the website a bit! I cannot find the HTML and JS in the code though ... I searched the whole project for the word `banner` (used as html-id for the logo) but nothing showed up.

I don't know if it is possible, but could you extract the frontend / html / css / js stuff from the source code so that more people could work on the project with less interference?

Cheers!
Tom

vitotai, 2017-07-25:
In fact, you can get the HTML/JS source by the browser.
However, that's compressed by shorten variable names.
The sources are "compressed", gzipped, and then convert to binary(hex) array, and put in data_xxx.c files. By doing this, a few tens of kilo bytes are saved.
The original source contains some comment and a lot of debugging code, like console.log(). It's better for developer, though. I can push these file if you need them.

tommueller, 2017-07-25:
>  I can push these file if you need them.
That'd be great!

vitotai, 2017-07-25:
In case you don't know. You can upload the file to "overwrite" the embedded one.
https://github.com/vitotai/BrewPiLess#development-tools

tommueller, 2017-07-26:
Ah, well that is easy as hell then :)
Would be great if you could provide the source-files, then I would get working on a alternative frontend.
Cheers!

allthepies, 2017-07-26:
Any clever stuff you can do with CSS to reduce the page sizes would be useful tom :)

tommueller, 2017-07-26:
I will get into it in 2 weeks, got some holiday to spend first ;)

vitotai, 2017-07-27:
Looking forward to it.
I am very bad at user interface, so I avoid creating one.

Heads-up:
ESP8266 will accept maximum 5 connections at the same time. Therefore avoid additional files. This is the reason that I merged all CSS, Javascript into the HTML file.

tommueller, 2017-08-11:
Sorry, my schedule is pretty tight these days, but I finally got 'round to do the first steps. I build a little grunt scaffolding around the code so that now files are split and building for development and production are automated. Just the minification you referred to is missing, maybe I can add that to.
If anyone wants to use that for their own development you can see it here: https://github.com/tommueller/BrewPiLess/commit/d0d87f02d14791e8eeceb0c3ccd8a8b0d62bbce9 or I can also of course send a PR. So far just organizing, no styling ;)

tommueller, 2017-08-11:
> Any clever stuff you can do with CSS to reduce the page sizes would be useful tom :)
Compared to the JS the CSS is really small already. Not so sure if I will dive into the JS, I think I will stick to changing the HTML and CSS for better styling ...

tommueller, 2017-08-12:
I am really not an expert in Arduino etc. development and was happy to use the prebuild bins of the library. I would like to test my updated files, but there is no binary with filemanager-support, right? Could you maybe also build out that version for poor frontend guys like me? Cheers

vitotai, 2017-08-12:
Filemanager is default supported except for SONOFF.

tommueller, 2017-08-12:
Ah sorry, I did not realize that I had to use the port 8008 when entering the filemanager. Alright, my grunt scaffolding seems to be working, just uploaded the current version of today and the file seems to be fine. Here is a little WIP screenshot.

I will close this issue since it is resolved and start a new discussion to get peoples opinion on the ui. Should I send a PR or rather start a seperate repository with the ui files and people can optionally choose to upgrade to my version?

closed 2017-08-12 by tommueller