vitotai/BrewPiLess #21: Frontend development discussion - input needed

tommueller, 2017-08-12:
I started working on a facelift for BrewPiLess' frontend. @vitotai added the files to the repository, see for this #18. The latest code can be found in my fork of the repository: https://github.com/tommueller/BrewPiLess

Because I don't have a lot of experience in being a BrewPiLess user I do need some input from users concerning the UI elements. Right now I am especially wondering about: 
- which are the most important information?
  - the chart
  - the gravity pane in the top
  - the menu in the bottom
- should the two parts of the page (plot and the settings on the bottom) stay on one page, or can they be split?
- any other ideas, suggestions?

I already cleaned up the code a little bit and started to make a clean, simple, material design layout which looks like this right now:

So please, any input is welcome!

vitotai, 2017-08-13:
That looks cool. Really.

In addition to mimicking the original BrewPi, another reason to put temperature control part under the chart is that it informative when beer profile is used. The beer profile chart and table provide the information of temperature schedule. (It's not pretty. I know.)

The LCD is refreshed every 5 seconds while the chart is updated every minute. The LCD also provides some other information about the system, like temperature control mode and idle/cooling/heating time.   A small piece of Javascript can used to retrieve the data, so that the information can be shown in other form.

The "iSpindel name" is not necessary, IMO. It's there because it's available.

zandoga, 2017-08-13:
Looks very good! I could not figure out what "state" is used for.

vitotai, 2017-08-13:
When the mouse cursor hovers over the line of chart, "state" will show the state of BrewPi, like "Idle", "Cooling", "Heating", "Prepare to Cool", and etc.

tommueller, 2017-08-14:
Thanks for the kind words.

The LCD was not meant to be chucked, I simply did not find a good spot to put it so far :)

> When the mouse cursor hovers over the line of chart, "state" will show the state of BrewPi, like "Idle", "Cooling", "Heating", "Prepare to Cool", and etc.

what exactly do you hover and where does it show?

tommueller, 2017-08-14:
Any chance that you have a local version of the server so that I can test the js as well?

vitotai, 2017-08-14:
> what exactly do you hover and where does it show?
When the mouse cursor hovers over the lines in the chart, the "state" will be REPLACED but the state string.

> Any chance that you have a local version of the server so that I can test the js as well?
I did use some simple PHP scripts to test the JS, but they are not maintained.
The original BrewPi core uses JSON format with an additional character at the beginning of output. bwf.js is used to be the interface between Javascript and BrewPi core. The purpose of testcmd.htm is to test the response of BrewPi core. (Later, I found it is useful to set additional setting and put it in the release code.)
I tried to shift the loading to the brewer, so most of the data is in JSON format and ESP8266 just pass what it is to JS mot of the time. 

Chart data is an exception for reduction of network traffic and processing of data by ESP8266. It is still simple in the JS part. Here is my PHP script to pass a saved log to JS.( It was quite dirty and complicated when I was developing the code for periodical chart updating. However, I don't even keep the code after finishing.)

`<?php
header('Content-Type: application/octet-stream');
header('Cache-Control: no-cache'); // recommended

$pmtime=0;
$linenum=3;
if(array_key_exists("offset",$_GET)){
	$offset=$_GET["offset"];
}else{
	$offset=0;
}

$filename ="coldsnap0509";

if(filesize($filename) > $offset){
$bin=fopen($filename,"r");
if($offset > 0){
	fseek($bin,$offset);
	echo fread($bin,filesize($filename)-$offset);
}else{
	echo fread($bin,filesize($filename));
}
}
?>`

tommueller, 2017-08-15:
Ok thanks!
One more question, does the chart have to stay at a fixed width of 800px, or can the script handle dynamic sizes?
Cheers

allthepies, 2017-08-15:
The recording div is populated when logging is enabled.  The logging filename is shown.

tommueller, 2017-08-15:
Ok, this got some progress today. You can test the current version if you download the HTML from my fork. It's still very much work in progress, I am just doing the chart in the first run. I figured out that it can have any size and __all the question asked so far, are now clear__.

I do have one more question though. What is shown as `fgstate`?

vitotai, 2017-08-15:
That is a small LED indicating the stability of gravity reading.
     0: "red",
     12: "orange",
      24: "yellow",
      48: "green"

The numbers denote the number of hours during which the gravity changes only a little, default to 1.

vitotai, 2017-08-20:
I just add a indicator for WiFi signal.

[image]

tommueller, 2017-08-21:
Cool, I will have some time over the upcoming weekend / beginning of next week and hope to be able to have a first running version ready then.

tommueller, 2017-08-28:
Some progress, will need a tester soon who has a version running with more then one temp sensor like me :) anyone?

allthepies, 2017-08-28:
I'll give it a go, will try on my testbed and then on my fridge setup which has three sensors, beer, fridge and room.

Do I just build/upload your fork using platformio or do I need to do some magic first ?

tommueller, 2017-08-28:
@allthepies great thanks. Gimme 2 more days, I'll let you know when it's done. You just need download the html from the dist folder and upload it using the filemanager.

allthepies, 2017-08-28:
Wilko.

vitotai, 2017-08-28:
I will test it, too. 
BTW, some people seems to like use their own names at the place of BrewPiLess. Therefore, I display the "hostname" there. It seems better for people who have more than one setup like me.

[image]

tommueller, 2017-09-01:
Ok, I pushed a first running version into my fork. I split up the main-page into 2 (chart and control), which creates some problems (dependencies between control part and chart). In order to have all the functions you need to upload both html-files from the dist folder.

Some notes:
- the lcd-text format seems to be set in the ESP8266 already? I was not able to hack into it, to format the text nicer for the new page-style
- hovering the charts shows some kind of popup? I haven't found the time to track the corresponding css, so right now hovering the chart makes the page a bit jumpy (because the "popup" is just appended to the page, scroll down to see the content)
- the control-page does not seem to work, because of some interdepencies with the chart. is that necessary? or is it okay to split the parts?

here is a screenshot for the ones not wnating to test it:
[image]

Looking forward to some feedback and maybe some help on fixing the open issues.

allthepies, 2017-09-01:
Looks good Tom, I'll give it a go over the w/e and report back.

Just so I understand, installation involves uploading the two html files from your repo htmljs/dist to a BPL instance using the BPL file manager ?

tommueller, 2017-09-01:
Cheers!

> Just so I understand, installation involves uploading the two html files from your repo htmljs/dist to a BPL instance using the BPL file manager ?

Exactly!

allthepies, 2017-09-01:
OK, got it up and running, looks great!

Initial comments below.

- Chart controls - toggle colours don't match the chart line colours, confusing.
Control page - Javascript error, tabs don't work
Uncaught TypeError: Cannot set property 'onmouseover' of null
    at init (control.html:1)
    at onload (control.html:1)
	
Control page - version different to that shown on main page  (main:  v2.3.2, control: v2.1)
Control page:  Shows "BrewPiless" banner rather than the hostname as per the main page.
Control page:  Colour mismatch, banner green, control pane is the original blue, could do with harmonising.

Navigation - have to explicitly open each sub-page in new tab otherwise main page is switched.  UI debate and I know magic new tabs is frowned upon, one for discussion.
Main page:  No wifi strength shown, just "wifi: "
Main page: When logging enabled, log filename is shown under the wifi strength but just the name.  Prefix with "Log filename:" or similar.

vitotai, 2017-09-01:
The LCD information can be extracted, and display in other way. A few regular expression would get the needed information from the plain text.

How about we break down the information into these fields:
BeerTemp
BeerSet
FridgeTemp
FridgeSet
RoomTemp
ControlMode
ControlState 
ControlStateSince ( the "time" parts)

Given these words are common, I would like to use a prefix, "lcd". How do you think?

tommueller, 2017-09-01:
@allthepies thanks for testing and the feedback, some comments below:

> Chart controls - toggle colours don't match the chart line colours, confusing.

I will look into this.

> Control page - Javascript error, tabs don't work
Uncaught TypeError: Cannot set property 'onmouseover' of null
at init (control.html:1)
at onload (control.html:1)

Control page - version different to that shown on main page (main: v2.3.2, control: v2.1)
Control page: Shows "BrewPiless" banner rather than the hostname as per the main page.
Control page: Colour mismatch, banner green, control pane is the original blue, could do with harmonising.

The control page is not working because it has some dependencies on the chart. Since I never used it I am not sure if this is a useful feature or if I can remove the code causing it. I need some input from you guys for this.

>  Navigation - have to explicitly open each sub-page in new tab otherwise main page is switched. UI debate and I know magic new tabs is frowned upon, one for discussion.

My plan was to integrate the other pages into the same header, footer style and have page changes, because I don't like opening tabs, but this can/should be debated here!

> Main page: No wifi strength shown, just "wifi: "

did you upgrade to version 2.3.2? You need to have the latest version to support this.

> Main page: When logging enabled, log filename is shown under the wifi strength but just the name. Prefix with "Log filename:" or similar. 

> Main page: Gravity axis value issue, see image below, the top gravity reading shows 0.54 which is less that than the value below it.

I don't think this a relayout issue, but rather an issue in the graph library, @vitotai ? I did not change anything on the graph itself.

tommueller, 2017-09-01:
> How about we break down the information into these fields:
BeerTemp
BeerSet
FridgeTemp
FridgeSet
RoomTemp
ControlMode
ControlState
ControlStateSince ( the "time" parts)

> Given these words are common, I would like to use a prefix, "lcd". How do you think?

@vitotai yes, sounds perfect! I am not 100% sure if the regex stuff should be done within the js-code or if it is in the BPL code, need some directions on what to do here, because I did not find where the lcd-text is generated.

allthepies, 2017-09-01:
I tested with the latest version i.e. 2,3.2  ( I always adopt new versions quickly after testbed proving :) )

re: control page, that is the key page for BPL.  Without it we cannot change the temp settings  (well I can with my Alexa skill but that's a mute point ;) )

vitotai, 2017-09-01:
The firmware part keeps a copy of LCD display. The javascript periodically issue  "l" command to get it. There are other commands like "s" and "t" that can be used to retrieve mode and current temperatures. However, the information of last line is only available on the LCD, and the response of LCD. I prefer parsing the LCD text to extract the information since it's simple enough, and using one command is better than using two. 

One issue remains. The last line of LCD is the status of the system, it can be the following patterns.
Idle: Idling for  %dh%02dm%02d
Cooling: Cooling for %dh%02dm%02d
Heating: Heating for %dh%02dm%02d
Waiting_to_cool: Wait to Cool %dh%02dm%02d
Waiting_to_cool:Wait to heat %dh%02dm%02d
cooling_min_time: Cool Time Left %dh%02dm%02d
heating_min_time: Heat Time left %dh%02dm%02d
Waiting_for_peak_detect: Waiting for peak
Door Open
Temp. control OFF
Unknown status!

Three options I have thought:
1. keep the whole sentence. Use the whole line as one.
2. Extract the time and state, for example
Idling for  %dh%02dm%02d =>  ControlState = "Idling" , ControlStateSince = "%dh%02dm%02d"
3.Using the state index. e.g., idle is 0, cooling is 4. Defined as variable "STATES".
Idling for  %dh%02dm%02d =>  ControlState = 0, ControlStateSince = "%dh%02dm%02d"

Your opinion?

vitotai, 2017-09-01:
I add the parsing function and modify the HTML to demonstrate.
[index.htm.zip](https://github.com/vitotai/BrewPiLess/files/1271724/index.htm.zip)

tommueller, 2017-09-02:
@vitotai and others: before I can continue working on this (and especially the control-page) we need to decide if we
a) need to have graph and control part on one page? and
b) want a "normal" navigation or should logs/config etc. pages open in a new tab as it is right now?

My personal opinions are:
for a) split control and graph pages, because it is different functionalities
for b) I prefer to stay in a tab and navigate between pages, this is really just what I expect from a page in 2017, especially because I do not need to have the chart open when I change something in the setup

What are your opinions?

zandoga, 2017-09-02:
You suggestion sounds good to me.

tommueller, 2017-09-02:
@vitotai the parsing works like a charm! already included it into my fork. thanks!

I updated my repo with a new version, I think all the issues on the graph-page should be resolved now, but please test again everyone!

@all please give more opinions on [my question before](https://github.com/vitotai/BrewPiLess/issues/21#issuecomment-326736597)

allthepies, 2017-09-02:
Hi Tom,

Just uploaded the files from your latest commit and I can see no difference.  What was supposed to have changed ?

Cleared browser several times etc.

tommueller, 2017-09-02:
@allthepies sounds like a cache problem. did you reload with shift pressed? maybe just try again (just pushed a new version again, with a little fix).

Changes are:
- better sorting and display of information in the sidepanel, including log-filename
- fixed wifi-colors, they were confusing
- toggle colors should match the line color in graph now
- some minor style improvements

allthepies, 2017-09-02:
I don't know what's going on Tom, it now refuses to load your files from the filesystem and is presenting the old pages from the program memory.  I've wiped the ESP8266 SPIFFS filesystem and started again but the same result.

allthepies, 2017-09-02:
OK!  

Your index.html file has to be uploaded as index.htm  (no trailing "l") as that's what BPL looks for in the SPIFFS filesystem.   Your control.html file has to be uploaded as-is  i.e. control.html   as that's what the main page HTML links to!

Confusing eh.  Perhaps rename index.html -> index.htm and change control.html page and reference links to control.htm ?

tommueller, 2017-09-02:
That is strange, I've always uploaded it as .html and it worked just fine. It doesn't feel like .htm is a valid file extension to me, so I changed it to html, but maybe I'll change it back ;)

Good that is working now! Thanks for testing and reporting back that issue!

@vitotai is there any reason for the .htm-file-ending? why is it working for me with .htm__l__ but not for @allthepies ? any ideas?

allthepies, 2017-09-02:
Lots of JS errors referencing LCD text

[image]

tommueller, 2017-09-02:
Mmh. That's weird. Can you try the attached file, I added some output to the console to see where the error comes from, cannot recreate it here. You have to rename it to index.htm of course.
[index.txt](https://github.com/vitotai/BrewPiLess/files/1272280/index.txt)

allthepies, 2017-09-02:
[image]

vitotai, 2017-09-02:
1. The index file should be "index.htm", I am not sure if "index.html" works or not. I've played with it for a while, and decide to use index.htm. Just forget why. Or maybe there isn't any reason.

2. ESP8266 or the upload program seems to have trouble handling BIG file. I was stuck at the issue for a while when I used AsyncWebServer. The default synchronized web server seems to work better, but it still has this issue. The symptom is the corrupted file. **You need to "reload" again. Or upload the file again.** It might be caused by upload or download.
Files smaller than 80k are safer. I never succeed on files over 130k.

tommueller, 2017-09-02:
@vitotai great thanks for the input. I renamed files in the output folder to .htm! Let me know when compression is an option, then I'll update the grunt-task to autocompress the files.

Do you have any idea why the parsing of the lcd-text is failing for @allthepies ? It works like a charm for me ... Don't know what's failing for him, but I only have one temperature (room), maybe it's an issue with other values?

allthepies, 2017-09-02:
My testbed has no temperature probes attached, perhaps that's the gotcha ?

vitotai, 2017-09-02:
I know the cause of JS error. I assumed the time is alway in %dh%02dm%02d, it turns out the "hour" doesn't exist. Give me a few minutes to verify the corrected one.

vitotai, 2017-09-02:
[index.htm.gz](https://github.com/vitotai/BrewPiLess/files/1272310/index.htm.gz)

The regular expressions are updated. Some more exception protection is also added.
The gzip is supported now in the source. (not pre-built images)
I recommend using highest compression ration, -9. so
`
gzip -9 index.htm
`
and just upload the index.htm.gz and let it be exact the name 'index.htm.gz' .

tommueller, 2017-09-02:
@vitotai sorry cannot find the changes in your file (I use seperate files for development and merge them in the end, so I cannot just override with your file). Can you maybe point me to them?
Also gzip will only be working once I updated my BPL, right? I only use your bins, so I'll have to wait for that.

allthepies, 2017-09-02:
Yes, you need the BPL version Vito's just commited to support SPIFFS served gzip files.

I've updated my test bed and am ready whenever you are...

vitotai, 2017-09-02:
You need new image to support GZ files. 

Without the gzip functionality, you have to use plain index.htm, not index.htm.gz.

tommueller, 2017-09-02:
@vitotai thanks for the bin

I do get the error now myself, but I cannot find the changes in your uploaded .htm file. Can you please post the differences? Or the functions that I need to override?

Thanks!

vitotai, 2017-09-02:
[index.htm.gz](https://github.com/vitotai/BrewPiLess/files/1272383/index.htm.gz)
I used old file. 

The updated functions here:
`

        function parseLcdText(lines) {
            var status = {};
            var modePatterns = {
                b: /Mode\s+Beer\s+Const/i,
                f: /Mode\s+Fridge\s+Const/i,
                p: /Mode\s+Beer\s+Profile/i,
                o: /Mode\s+Off/i
            };
            var modes = Object.keys(modePatterns);
            status.ControlMode = "i";
            for (var m = 0; m < modes.length; m++) {
                if (modePatterns[modes[m]].test(lines[0])) {
                    status.ControlMode = modes[m];
                    break;
                }
            }
            status.ControlState = i;
            var tempRE = /\s*(\w+)\s+(.+)\s+(.+)\s+.+([CF])\s*$/;
            for (var i = 1; i < 3; i++) {
                var temps = tempRE.exec(lines[i]);
                status[temps[1] + "Temp"] = temps[2];
                status[temps[1] + "Set"] = temps[3];
                status.format = temps[4];
            }
            var i = 0;
            var statePatterns = [
                /Idling\s+for\s+(\S+)\s*$/i,
                /control\s+OFF/i,
                /Door\s+Open/i,
                /Heating\s+for\s+(\S+)\s*$/i,
                /Cooling\s+for\s+(\S+)\s*$/i,
                /Wait\s+to\s+Cool\s+(\S+)\s*$/i,
                /Wait\s+to\s+Heat\s+(\S+)\s*$/i,
                /Wait\s+for\s+Peak/i,
                /Cool\s+Time\s+left\s+(\S+)\s*$/i,
                /Heat\s+Time\s+left\s+(\S+)\s*$/i
            ];
            status.ControlStateSince = "";
            for (i = 0; i < statePatterns.length; i++) {
                var match = statePatterns[i].exec(lines[3]);
                if (match) {
                    if (typeof match[1] !== "undefined") status.ControlStateSince = match[1];
                    break;
                }
            }
            status.ControlState = i;
            status.StatusLine = lines[3];
            return status;
        }

        function processLcdText(lines) {
            Q("#status-pane .error").style.display = "none";
            var status = parseLcdText(lines);
            var ModeString = {
                o: "OFF",
                b: "Beer Constant",
                f: "Fridge Const",
                p: "Beer Profile",
                i: "Invalid"
            };
            Object.keys(status).map(function(key, i) {
                var div = Q("#lcd" + key);
                if (div) {
                    if (key == "ControlMode") div.innerHTML = ModeString[status[key]];
                    else if (key == "ControlState") div.innerHTML = (status[key] < STATES.length) ? STATES[status[key]].text : "Unknown State";
                    else div.innerHTML = status[key];
                }
            });
        }
`

tommueller, 2017-09-02:
Thanks @vitotai !

I fixed the code in my branch and it should be ready for testing @allthepies

allthepies, 2017-09-02:
:+1: 

Temps now displaying OK and no lcd console error.

However, hovering over the Wifi singnal strength indicator gives a JS error (see image).  Should display the %age strength.

[image]

tommueller, 2017-09-02:
Ah well spotted, the mouseover should still work though, the value is added as HTML-title attribute now, so it's popping up with a delay of about 1s!

I removed the lines that caused the error and fix is pushed. Thanks

tommueller, 2017-09-03:
Allright, did quite some work this morning to clean up and split up the scripts and start to get the control page done. I have just pushed a new version, please test the index and also the control, some notes:

- graph-page (index.htm) should be almost done by now! I fixed some smaller issues, added ui-validation for gravity-input, styled the inputs etc. please have a go to try and test all the functionality!
- I started working on the control-page
  - it loads hostname and version from BPL now
  - added general styles, replaced the tabs with better looking ones
  - I moved around quite some scripts, to get a cleaner and smaller file, so please test funcationality!
  - @vitotai when I am on the control-page the lcd-writing apparently always sends "Controller not updating data" back, which I show in a notification on the bottom, because I treat it as an error. Why is this and should I just dismiss the notification or is there a problem with the script?

tommueller, 2017-09-03:
Okay. Control-Page is now also done, as far as I can tell. So please: testing, testing, testing! And let me know if you don't like any of the applied styles!
Cheers!

vitotai, 2017-09-03:
> when I am on the control-page the lcd-writing apparently always sends "Controller not updating data" back, which I show in a notification on the bottom, because I treat it as an error. Why is this and should I just dismiss the notification or is there a problem with the script?

The init() function were created by merging control and graph pages. To update the LCD, the script periodically issuing "l" command. The script also check if any response from the server. It is not necessary for pages which don't have LCD.

`
function init() {
  loaded();
  var gotMsg=true;

  BWF.init({
   /* this is not necessary
    onconnect:function(){
      BWF.send("l");
      setInterval(function(){
        if(!gotMsg) controllerError();
        BWF.send("l");
        gotMsg=false;
      },5000);
    }, */
    error:function(e){
      communicationError();
    },
    handlers:{
      V:function(c){
        if(typeof c["nn"]!="undefined"){
          Q("#hostname").innerHTML = c["nn"];
        }
        if(typeof c["ver"]!="undefined"){
          if(JSVERION != c["ver"]) alert("Version Mismatched!. Reload the page.");
          Q("#verinfo").innerHTML = "v" + c["ver"];
        }
      }
    }
  });
};
`

tommueller, 2017-09-03:
Yes, I put this in to have hostname and version info available to put in in the header. But I see which part I can get rid of, thanks!

vitotai, 2017-09-03:
It was over 100 Fahrenheit yesterday, and I found the regex will fail if Fahrenheit is used. Please update the regex. 

`
  var tempRE = /\s*([a-zA-Z]+)\s*([-\d\.]+)\s+([-\d\.]+)\s+(\S+[CF])\s*$/i;
`

BTW, how do think about adding the temperature unit?
`
 else if (key.match(/[Temp|Set]$/)) div.innerHTML = status[key] + status.tempUnit;
`

The full code:

`
        function parseLcdText(lines) {
            var status = {};
            var modePatterns = {
                b: /Mode\s+Beer\s+Const/i,
                f: /Mode\s+Fridge\s+Const/i,
                p: /Mode\s+Beer\s+Profile/i,
                o: /Mode\s+Off/i
            };
            var modes = Object.keys(modePatterns);
            status.ControlMode = "i";
            for (var m = 0; m < modes.length; m++) {
                if (modePatterns[modes[m]].test(lines[0])) {
                    status.ControlMode = modes[m];
                    break;
                }
            }
            status.ControlState = i;
            var tempRE = /\s*([a-zA-Z]+)\s*([-\d\.]+)\s+([-\d\.]+)\s+(\S+[CF])\s*$/i;
            for (var i = 1; i < 3; i++) {
                var temps = tempRE.exec(lines[i]);
                status[temps[1] + "Temp"] = temps[2];
                status[temps[1] + "Set"] = temps[3];
                status.tempUnit = temps[4];
            }
            var i = 0;
            var statePatterns = [
                /Idling\s+for\s+(\S+)\s*$/i,
                /control\s+OFF/i,
                /Door\s+Open/i,
                /Heating\s+for\s+(\S+)\s*$/i,
                /Cooling\s+for\s+(\S+)\s*$/i,
                /Wait\s+to\s+Cool\s+(\S+)\s*$/i,
                /Wait\s+to\s+Heat\s+(\S+)\s*$/i,
                /Wait\s+for\s+Peak/i,
                /Cool\s+Time\s+left\s+(\S+)\s*$/i,
                /Heat\s+Time\s+left\s+(\S+)\s*$/i
            ];
            status.ControlStateSince = "";
            for (i = 0; i < statePatterns.length; i++) {
                var match = statePatterns[i].exec(lines[3]);
                if (match) {
                    if (typeof match[1] !== "undefined") status.ControlStateSince = match[1];
                    break;
                }
            }
            status.ControlState = i;
            status.StatusLine = lines[3];
            return status;
        }

        function processLcdText(lines) {
            Q("#status-pane .error").style.display = "none";
            var status = parseLcdText(lines);
            var ModeString = {
                o: "OFF",
                b: "Beer Constant",
                f: "Fridge Const",
                p: "Beer Profile",
                i: "Invalid"
            };
            Object.keys(status).map(function(key, i) {
                var div = Q("#lcd" + key);
                if (div) {
                    if (key == "ControlMode") div.innerHTML = ModeString[status[key]];
                    else if (key == "ControlState") div.innerHTML = (status[key] < STATES.length) ? STATES[status[key]].text : "Unknown State";
                    else if (key.match(/[Temp|Set]$/)) div.innerHTML = status[key] + status.tempUnit;
                    else div.innerHTML = status[key];
                }
            });

        }

`

tommueller, 2017-09-08:
@vitotai great, thanks. fixed the issue fahrenheit issue and I think showing the temperature unit is a great idea, both is updated in the current version of my repo

tommueller, 2017-09-08:
I would love for some more people to test and then create a PR in a couple of days! I am also happy if people have suggestions on how to improve the layout!

gmoyle, 2017-09-08:
I'm interested in testing the new interface, I "thought" I understood how to put it on my brewpiless but no luck. Is there a bin I can just upload ota?

tommueller, 2017-09-08:
hey @gmoyle . Cool thanks!

There is no bin so far, but uploading is really easy:

- download the files from my fork: https://github.com/tommueller/BrewPiLess/tree/master/htmljs/dist
- go to the filemanager: brewpiless.local:8008/filemanager
- upload the files using the filemanager
  - select index.htm and upload it, do the same for control.htm
  - if you are running the bin that supports gzip [BrewPiLess.v2.3.2.gzip.bin](https://github.com/vitotai/BrewPiLess/blob/master/bins/BrewPiLess.v2.3.2.gzip.bin) you can upload
- reload the brewpiless page, hold the shift button pressed when clicking the button or use ctrl+shift+r to reload page without using the browsers cache

[62 more replies]

closed 2018-02-13 by tommueller