vitotai/BrewPiLess #22: Enhancement request:  Add webservice method

allthepies, 2017-08-23:
Vito,

I've written an Amazon Alexa skill to interact with BPL, it can get current beer temp, change beer set point etc.   It can also give a status report of BPL, but to do this I've had to add a webservice method to return a JSON object with current state/mode, beer set/tmp, fridge set/temp + room temp.

Would it be possible to merge this into your code ?   Small changes as below:

BrewPiProxy.cpp - add this method

`void BrewPiProxy::getAllStatus(uint8_t *pState,uint8_t *pMode,float *pBeerTemp,float *pBeerSet,float *pFridgeTemp, float *pFridgeSet, float *pRoomTemp)
{
	*pBeerTemp=temperatureFloatValue(tempControl.getBeerTemp());
	*pBeerSet=temperatureFloatValue(tempControl.getBeerSetting());
	*pFridgeTemp = temperatureFloatValue(tempControl.getFridgeTemp());
	*pFridgeSet = temperatureFloatValue(tempControl.getFridgeSetting());
	*pRoomTemp =temperatureFloatValue(tempControl.getRoomTemp());
	*pState = (uint8_t) tempControl.getState();
	*pMode = (uint8_t) tempControl.getMode();
}`

BrewPiLess.cpp - add the following

#define 

`#define GETSTATUS_PATH		"/getstatus"`

Code to handle the GETSTATUS_PATH Get request and return JSON

`		}else if(request->method() == HTTP_GET && request->url() == GETSTATUS_PATH){
			uint8_t mode, state;
	 		float beerSet, beerTemp, fridgeTemp, fridgeSet, roomTemp;
	 		brewPi.getAllStatus(&state, &mode, &beerTemp, &beerSet, &fridgeTemp, &fridgeSet, &roomTemp);
	 		String json=String("{\"mode\":\"") + String((char) mode)
	 			+ String("\",\"state\":") + String(state)
	 			+ String(",\"beerSet\":") + String(beerSet)
				+ String(",\"beerTemp\":") + String(beerTemp)
				+ String(",\"fridgeSet\":") + String(fridgeSet)
				+ String(",\"fridgeTemp\":") + String(fridgeTemp)
				+ String(",\"roomTemp\":") + String(roomTemp)
				 +String("}");
	 		request->send(200,"application/json",json);

	 	}`

Add GETSTATUS_PATH to canHandle

`	bool canHandle(AsyncWebServerRequest *request){
	 	if(request->method() == HTTP_GET){
	 		if(request->url() == POLLING_PATH || request->url() == CONFIG_PATH || request->url() == TIME_PATH
	 		|| request->url() == RESETWIFI_PATH || request->url() == CONTROL_CC_PATH || request->url() == GETSTATUS_PATH
	 		#ifdef ENABLE_LOGGING
	 		|| request->url() == LOGGING_PATH
	 		#endif
	 		){
	 			return true;`

Obviously I can maintain all that in my own fork but would be good to integrate into the main codebase IMO.

universam1, 2017-08-23:
That's pretty amazing! How about Alexa skill?

allthepies, 2017-08-23:
The Alexa skill side is pretty easy.  Define the speech model and then write an AWS Lambda function which is invoked to implement all the identified actions (intents in Alexa speak).  All available free of charge in the Amazon free tier :)   My Lambda function just invokes webservice calls on BPL to set the beer temp, get the beer temp and get the overall BPL status (which is the part I had to add functions to BPL to achieve).

vitotai, 2017-08-23:
brewPi.getAllStatus should already there, right?

One question. What if the temperature is invalid? Should "null" be used or the invalid value, -1000.0?

allthepies, 2017-08-24:
Hi Vito,

I added getAllStatus  to BrewPiProxy.cpp (probably should be defined in the .h file too).  Everything in my post above is added.

Regarding invalid temp, null is probably best but I'll leave that up to you if you decide to implement.

Thanks!

vitotai, 2017-08-24:
[image]

The getAllStatus() is exact the function I used to get status in logging. I just wanted to make sure if there is something different between yours and mine. Please check the source.

allthepies, 2017-08-24:
Sorry yes!  The getAllStatus() method is yours.  I'm just using it to expose the data via the main BPL webservices interface.

So just need:

BrewPiLess.cpp - add the following

New #define for the new webservice

#define GETSTATUS_PATH "/getstatus"

Code to handle the GETSTATUS_PATH Get request and return JSON

`	}else if(request->method() == HTTP_GET && request->url() == GETSTATUS_PATH){
uint8_t mode, state;
float beerSet, beerTemp, fridgeTemp, fridgeSet, roomTemp;
brewPi.getAllStatus(&state, &mode, &beerTemp, &beerSet, &fridgeTemp, &fridgeSet, &roomTemp);
String json=String("{"mode":"") + String((char) mode)
+ String("","state":") + String(state)
+ String(","beerSet":") + String(beerSet)
+ String(","beerTemp":") + String(beerTemp)
+ String(","fridgeSet":") + String(fridgeSet)
+ String(","fridgeTemp":") + String(fridgeTemp)
+ String(","roomTemp":") + String(roomTemp)
+String("}");
request->send(200,"application/json",json);

 	}`

vitotai, 2017-08-24:
Would you like to share your Alexa skill?

allthepies, 2017-08-24:
I will do, just need to think of the best way of doing that.  The voice model is independent of the target BPL installation but the AWS Lambda function invoked by the voice model is very installation specific currently i.e. URLs for my BPL instance on the internet + authentication details hardcoded for the setting beer temp operation.   I could put a version up in github which has placeholders for all the installation specific stuff and leave it to the implementer to fill in the detail.

allthepies, 2017-08-24:
Thanks for implementing the getstatus operation !

allthepies, 2017-08-25:
Ah, I have found a little bug with the getstatus implementation.

The JSON returned is invalid and causes my Alexa Lambda app to error.

You're not enclosing the mode value (which is a char) with quotes in the JSON string.

"mode":b

should be 

"mode":"b"

Just needs a simple change

`String json=String("{\"mode\":\"") + String((char) mode)
			+ String("\",\"state\":") + String(state)`

And Alexa is happy again :+1:

allthepies, 2017-08-25:
Thanks!  

What is the EMI workaround thing BTW ?

vitotai, 2017-08-25:
oops. I made mistake again.

Sometimes, the LCD display might be scrambled because of EMI, Electromagnetic interference, usually caused by starting/stopping the compressor. The workaround is just re-initialization and re-printing the screen periodically, say 10 minutes.

allthepies, 2017-08-25:
No problems Vito, thanks for your flexibility and continued development of the project.

allthepies, 2017-08-28:
Alexa skill added to GitHub.  No instructions so far on how to configure it though!

I'll add those in due course but if anyone has constructed an Alexa skill before then the repo contents as-is should get you going.

The key thing to change is the fridgeAddress object values in src/constants.js as this identifies the internet accessible address of your BPL fridge.

https://github.com/allthepies/BrewPiLessAlexa

vitotai, 2017-08-28:
cool. Thanks.