vitotai/BrewPiLess #25: Frontend Decisions - Split functionalities and navigation

tommueller, 2017-09-02:
Since #21 became quite technical and messy, I open this issue to discuss two major UI decisions that we have to make.

a) Should the graph and control part be on one page or not? and
b) Should we have a "normal" navigation or should logs/config etc. pages open in a new tab as it is right now?

My personal opinions are:
for a) split control and graph pages, because it is different functionalities
for b) I prefer to stay in a tab and navigate between pages, this is really just what I expect from a page in 2017, especially because I do not need to have the chart open when I change something in the setup

What are your opinions?

allthepies, 2017-09-02:
My opinions:

a) Happy with split graph/control pages.
b) Happy with normal clickthrough behavior i.e. will open in the current tab unless the user uses browser capability to open in a new tab.

vitotai, 2017-09-02:
a) I separated the control with the main page at the beginning. However, When using beer profile, I feel like wanting to see the chart and beer profile at the same time. Merging the page caused me some pain, though. One of the biggest issue is security. Although there is only simple authentication available, the control page should be protected while the main page might not.
Separating the control and viewing pages should be a good option. 

b) I focus on functionalities and pay no attention to the flow or anything else. I just try my best not adding too much stuff and make the file size big.

tommueller, 2017-09-08:
Ok, so all the feedback seems to be that it is
a) ok, to seperate control from graph page and
b) normal clickthrough behavior is fine

@vitotai could you also add the missing pages (logs, setup etc.) so I can match them stylewise?

vitotai, 2017-09-08:
@tommueller 

Not all files work like the main page. Some are embedded in the code, and uploaded HTML files will never be looked up. 
Files that work like index.htm
lcd.htm : the small LCD display(only the LCD)
setup.htm: the setup
testcmd.htm: the interface to BrewPi core.

Pages that embedded in the code,
log - remote and local log configuration.
config - hostname, user/pass
gdc  - iSpindel setup

tommueller, 2017-09-11:
@vitotai well it should be able to override the three embedded pages in the code, right?

or are you not planning to implement the new ui as a new default at some point into the project?

vitotai, 2017-09-11:
@tommueller

I am reviewing the code, thinking maybe I should split all the HTML files and handle them in the same way. It can be easily done by copy and paste to another files, except the config. Appending "htm" postfix might be better, then.

BTW,
I seems to lose the original uncompressed setup.htm. Anyway, I will push them onto GitHub.

vitotai, 2017-09-12:
@tommueller 
I just updated. Now all files can be replaced by the html(ro htm.gz) at file system.

At some point, I might make your work standard option while keep the classic option, if you don't mind.

tommueller, 2017-09-12:
@vitotai sounds perfect. will do the rest of the pages probably on the weekend.

> At some point, I might make your work standard option while keep the classic option, if you don't mind.

sure, having options if always good ;)

tommueller, 2017-11-10:
This discussion is finished. Decisions have been made :)

closed 2017-11-10 by tommueller