vitotai/BrewPiLess #26: Hardware: are there any plans for a decent BPL case?

tommueller, 2017-09-08:
I would love to finish my BPL-build soon and use it to control my fridge. Currently this is done by an Inkbird. But I do want to have a decent case for the BPL first, as there are high-voltages going to the SSRs. Are there any plans available? Models for 3D-printers? Lasercutter?
Or maybe some people could just post some pics of their setups to inspire me ;)
Cheers
Tom

jalexandre0, 2017-09-10:
I´m using the PCB and cases from my old project.   It is a plastic case and  basic PCB. 
The only change is software. Dropped SACI and upload BrewPi Less. 

Some pictures: 
https://loja.brassar.com.br/saci-controlador-de-temperatura-web

Schematics: 
https://github.com/jalexandre0/saci-temperature-controller/tree/master/schematics

tommueller, 2017-09-11:
@jalexandre0 cool, that looks good!

Can you post a link to the coil relays? How many Watts can they cope with? Enough for a fridge and a heating device?

vitotai, 2017-09-11:
Robert from Sweden has designed a board.
http://www.homebrewtalk.com/showthread.php?t=596907&page=8

jalexandre0, 2017-09-14:
@tommueller  I use relays like these  5V and 10A-> 
https://pt.aliexpress.com/item/10Pcs-SRD-05VDC-SL-C-DC-5V-Rating-Coil-SPDT-Miniature-Power-Relay-Blue/32753270224.html

Most refrigerators have less than 2A of current. Heat element can vary. I take the simplest route and use a lamp. Cheap and effective. 

If you have any questions, just ask.

universam1, 2017-09-14:
Did someone evaluate the Sonoff devices already? There are various editions available that could be used as a good base. I ordered one, lets see.

tommueller, 2017-09-14:
@universam1 that is a great idea! Never used one of them, but they look perfect for the job. Looking forward to your comments on it.

vitotai, 2017-09-14:
Some information about SONOFF:
http://www.homebrewtalk.com/showthread.php?t=587425&page=25

Due to the small 1M flash size, the default configuration of SONOFF doesn't support
1. LCD
2. Web OTA update
3. File Manager
The space of file system is 512K.

To support OTA, double program space is needed. Current program size of BPL is around 450k, and that means at least 900k is needed to support OTA, which leaves only 128k or 64k file system space.

allthepies, 2017-09-14:
It is possible to swap out the flash chip for a larger one though :)

vitotai, 2017-09-14:
@allthepies 
You can.
https://www.youtube.com/watch?v=8mz5sCAvDAY
But I don't think I can do it. 
It's too difficult for me.

universam1, 2017-09-14:
There are different editions with various features available, do they all have the same memory chip?

vitotai, 2017-09-15:
> There are different editions with various features available, do they all have the same memory chip?

They don't list the flash size in the specifications, but I think they are all the same.

tommueller, 2017-09-15:
My thinking was to use 2 of the sonoff smart plugs (like this: https://www.amazon.de/VIGICA-Steckdose-intelligente-Fernsteuerung-Android/dp/B073GNFVDM/ref=sr_1_36?ie=UTF8&qid=1505463692&sr=8-36&keywords=sonoff+switch) and switch them on/off from BPL.

lekrom, 2017-09-17:
Dear @tommueller,
@allthepies started a google group to share some stuff regarding BPL builds:
https://groups.google.com/forum/#!forum/brewpiless
I posted my PCB design as well as my 3D-printed enclosure design if youre interrested.
-Lekrom

tommueller, 2017-09-17:
Hey @lekrom , that is great, thanks for the link!

tommueller, 2017-09-19:
Which Sonoff by the way are we talking about? I find many different versions ... Because they only have one socket, right? How can you then have a cooling and a heating device connected?

universam1, 2017-09-19:
take this one for example, the dual: http://www.exp-tech.de/itead-studio-sonoff-dual-wifi-wireless-smart-switch
And switching the flash chip is probably way easier than any other DIY attempt: http://www.andremiller.net/content/upgrading-sonoff-wireless-smart-switch-flash-memory-esp8266

vitotai, 2017-09-19:
I wonder why they don't make 4M bytes version.
Chinese manufacturers are usually flexible. Maybe they can accept a customized order for a quantity, say 1k.

quimnut, 2017-10-07:
I was reading the sonoff dual's relays are controlled by serial unfortunately. https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual

vitotai, 2017-10-07:
More detail information:
http://tinkerman.cat/esp8266-multi-relay-boards-sonoff-dual-electrodragon/

It's interesting that 2ch uses additional chip while 4ch uses GPIOs.

BrewPi supports SSR control over OneWire, OneWireActuator. I haven't tested it but the idea is basically the same for GPIO extensions. Just different buses, OneWire, I2C, and UART. It's possible to support that UART GPIO extender in the same way that OneWireActuator does.

tommueller, 2018-02-13:
I will close this, as the dicussion has been inactive for a while now.

closed 2018-02-13 by tommueller

Paiakan, 2018-05-21:
@tommueller I just finished two versions of cases for the BrewPiLess: One in 3D printing and one using acrylic.
https://www.thingiverse.com/thing:2924101
https://www.thingiverse.com/thing:2787182