vitotai/BrewPiLess #28: Trouble installing on Wemos D1 R2

oleost, 2017-09-13:
Hi

I have downloaded Arduino 1.8.2 and BrewPiLess-2.0.

I have added the ESP8266 board to Arduino IDE.
I have copied all of `BrewPiLess-2.0\BrewPiLess-2.0\libs\libs\libs` to my `Documents\Arduino\libraries` folder, I also fetched the OneWire library and manually placed it there as it seemed to be missing.

[image]

I up get to upload the sketch without any issues to my Wemos, but from there I`m stuck.
The Wifi indicator does not seem to light up, and after waiting more than 3 minutes a new hotspot never got created.

Any advice ?

vitotai, 2017-09-13:
I don't get it. What is "Wifi indicator"?
Initially, BrewPiLess will create a network named "brewpiless". You have to connect to that network and setup the network.

BTW, Arduino IDE was ditched. Use PlatformIO for newer version instead.

oleost, 2017-09-13:
I thought one of the leds was and indicator for traffic going on Wifi, but that was probably just another program I had laying on it earlier. So newer mind in regards to that.

But still no new network to connect to, even when trying the BrewPiLess-0.2.1 version.

But I will give PlatformIO a go, need to read up on it first, I saw you had switched, but I thought that was only in regards to writing BrewPiLess, and that you could still use Arduino IDE to flash without issues.

oleost, 2017-09-13:
Okey, So I now tried PlatformIO ( In Atom )
Downloaded BrewPiLess-master and changed platformio.ini to `board = d1_mini` ( Even though its an Wemos D1 R2 full size but I believe they use same target ? )

Got to build and upload it succesfully, but still no new network to connect to.

Is there something else I have to do ?

vitotai, 2017-09-14:
I don't know. 
If that specific board has ever connected to your network, it might already connects to your AP. That confidential is stored by ESP8266.

oleost, 2017-09-14:
Thank you, that was it. I want not aware that ESP8266 stored that information.

BTW, are there any other places to discuss BrewPiLess than here and your blog?
And are there some guide / info somewhere how to get started except the readme?

Also do you mind user support issue like this, or don't you want them?

closed 2017-09-16 by oleost