vitotai/BrewPiLess #29: Timezone question

lekrom, 2017-09-15:
Dear @vitotai ,

Probably a dumb question:
My BPL always seems to display the time as GMT timezone (Not where I am at), I noticed that the sntp code sets the timezone to 0. I tried to change this but it had no effect.
 Is there a way to detect / adjust according to the local timezone? The sntp library documentation is rather bad...

Regards,
Lekrom.

universam1, 2017-09-15:
Instead of NTP one could parse a header response from time.is which is a location discovery service. It repeats your local time.

vitotai, 2017-09-15:
@lekrom
You are talking about the 5th line on OLED display, don't you?
A easier way is adding the timezone offset before it is printed.
BrewPiLess.cpp:1264
`
makeTime(now,t);
`
"add" timezone offset like: (-8 hours in the example.)
`
makeTime(now -8 * 60* 60,t);
`

@universam1 
I tried time.is by curl and found out that HTTPS is needed while time.is forbids automatic access.
I had thought of getting the timezone information from browsers. The 5th line display is only available on OLED, so I don't bother doing that before I touch that part of code again.

lekrom, 2017-09-15:
Dear @vitotai,

Indeed, on the Oled. After I reported the issue I noticed the browser displays it correctly. I will try your workaround. Thanks.
Finally got all my parts (ebay and alliexpress sometimes slow in my part of the world) to build my BPL.
Busy testing it all out to make sure my homemade PCB works reliably.. I will do some pawn-shop trolling tomorrow looking for a good deal on a chest-freezer. Too hot (at the moment) to ferment in South Africa without it...
Again, Epic project, Big Fan of your work...

-Lekrom

allthepies, 2017-09-15:
Getting a good community together now Vito.

Worth creating a google group for BPL ?

General Q&A + build stories/pics etc.  Perhaps a better forum than Githib issues ?

https://groups.google.com/forum/#!overview

lekrom, 2017-09-15:
@vitotai 
Sweeet. Your suggestions worked.
If you want to start the group as @allthepies  suggested, I'm in. Busy printing my enclosure, designed and built a PCB. I'd love to share if anyone is interested...

-Lekrom

lekrom, 2017-09-16:
Hi @vitotai 

I thought I'd share my implementation of your workaround:

BrewPiLess.cpp:1264:
#ifdef TIMEZONE
   makeTime(now + TIMEZONE *60*60,t);
#else
  makeTime(now,t);	
#endif

This way you can set the timezone by using a build-flag. A bit more flexible I think...

allthepies, 2017-09-16:
Lekrom, I'd love to see your enclosure and PCB designs.

I've also created a BPL google group where this kind of information can be shared.

https://groups.google.com/forum/#!forum/brewpiless

lekrom, 2017-09-16:
Hi @allthepies,  
I will share shortly, have to make a few minor tweaks.
Joined the group.
-Lekrom

vitotai, 2017-09-16:
@lekrom
For people who live in country that uses day light saving, that won't work well.

My idea is to use the information from browser, and let Javascript code set the timezone automatically.

lekrom, 2017-09-16:
I agree. A better sollution, more robust. We do not use DST, so I am ok either way...

vitotai, 2017-09-17:
FYI, 
I've done it when I was roasting coffee yesterday.

closed 2017-11-06 by lekrom