vitotai/BrewPiLess #32: Default Settings

gmoyle, 2017-09-21:
I have been trying to really fine tune my pid values and other settings. I realized that I did not know how to reset any of the defaults.

I was wondering if under the JSON section of the documentation if you could add a column DEFAULTS and list what the default setting is for each of the defined options? 

Or are they listed somewhere else?

Thanks for putting so much time into this. I am really loving it!

vitotai, 2017-09-22:
The default setting is set by ERASING the EEPROM, which is usually done when you tried to setup the devices.

gmoyle, 2017-09-22:
Yes technically that is true, but truly a nuclear option. If I have messed with kp, kd, ki settings and I want to put them back to defaults, or a few of the other options, Erasing the EEPROM would be more work than using the test page to just put a few settings back to defaults. 

I just thought listing them would be a nice documentation touch and helpful to end users.

vitotai, 2017-09-29:
The effect of ERASHING EEPROM is exact resetting to default. It is exact the term that original BrewPi uses. I agree with you that ERASHING seems terrifying.