vitotai/BrewPiLess #33: Unable to compile in PlatformIO

stagma, 2017-09-22:
Hi,

I have changed the following in config.h = '#define BREWPI_ROTARY_ENCODER 1'
I Have also changed the following in espconfig.h = '#define EnableGravitySchedule false'

But everytime i try to build the the project I get the following result:

`In file included from src/BrewKeeper.cpp:6:0:
src/BrewKeeper.h: In member function 'void BrewKeeper::setStableThreshold(uint8_t)':
src/BrewKeeper.h:120:54: error: 'class BrewProfile' has no member named 'setStableThreshold'
void setStableThreshold(uint8_t threshold){_profile.setStableThreshold(threshold);}
^
src/BrewKeeper.cpp: In member function 'bool BrewProfile::load(String)':
src/BrewKeeper.cpp:471:31: error: 'PROFILE_JSON_BUFFER_SIZE' was not declared in this scope
DynamicJsonBuffer jsonBuffer(PROFILE_JSON_BUFFER_SIZE);
^
*** [.pioenvs/nodemcuv2/src/BrewKeeper.o] Error 1
In file included from src/BrewPiLess.cpp:40:0:
src/BrewKeeper.h: In member function 'void BrewKeeper::setStableThreshold(uint8_t)':
src/BrewKeeper.h:120:54: error: 'class BrewProfile' has no member named 'setStableThreshold'
void setStableThreshold(uint8_t threshold){_profile.setStableThreshold(threshold);}
^
In file included from src/BrewPiLess.cpp:57:0:
src/ExternalData.h: In member function 'void ExternalData::setGravity(float, time_t)':
src/ExternalData.h:195:28: error: 'IsGravityValid' was not declared in this scope
if(!IsGravityValid(old_sg)) filter.setInitial(sg);
^
In file included from src/ExternalData.h:4:0,
from src/DataLogger.cpp:12:
src/BrewKeeper.h: In member function 'void BrewKeeper::setStableThreshold(uint8_t)':
src/BrewKeeper.h:120:54: error: 'class BrewProfile' has no member named 'setStableThreshold'
void setStableThreshold(uint8_t threshold){_profile.setStableThreshold(threshold);}
^
In file included from src/DataLogger.cpp:12:0:
src/ExternalData.h: In member function 'void ExternalData::setGravity(float, time_t)':
src/ExternalData.h:195:28: error: 'IsGravityValid' was not declared in this scope
if(!IsGravityValid(old_sg)) filter.setInitial(sg);
^
src/DataLogger.cpp: In member function 'int DataLogger::dataSprintf(char*, const char*)':
src/DataLogger.cpp:59:52: error: 'IsGravityValid' was not declared in this scope
d += printFloat(buffer+d,sg,3,IsGravityValid(sg));
^
*** [.pioenvs/nodemcuv2/src/DataLogger.o] Error 1
*** [.pioenvs/nodemcuv2/src/BrewPiLess.o] Error 1`

I'm unsure what I can change from here to make it compile.

allthepies, 2017-09-22:
Does it compile without your changes ?  i.e. with the version direct from github.

vitotai, 2017-09-22:
The option might not work as expected.
#define EnableGravitySchedule false

I haven't check it for a while.

stagma, 2017-09-22:
so just changed the option "EnableGravitySchedule" back to true and I can compile it.

Maybe I just have to build an iSpindle... but is there anything I can do in the meantime so i can run it standalone?

allthepies, 2017-09-23:
What do you perceive the problem to be ?  I run BrewPiless without an iSpindel, I ignore the gravity side of the application in fact.  It will work standalone with that flag set to true.

stagma, 2017-10-04:
Thanks for that, I have complied it with flag as true and will do further testing once this beer has finished fermenting. I just wasn't sure if you were able to ignore gravity or not.

closed 2017-10-04 by stagma