vitotai/BrewPiLess #35: Main page constantly loading

sahjhan, 2017-09-24:
Just flashed a esp8266 with version 2.3.3 with the provided bin. After connecting it to my AP I get a constantly loading blank page. I can access the /config /setup pages without issue but the main page /lcd and /log will not load. just a constant page loading and blank screen.

allthepies, 2017-09-25:
Which browser/OS + ESP8266 variant (NodeMCU, Wemos D1 etc) are you using ?  And is your browser connected to the same AP as BrewPiLess directly ?  i.e. not going via mobile.

Try clearing your browser cache and then retry.  If still no joy then access the developer console of the browser and look for any reported errors.

Also, if you access the /fs   URL do you get a sequence of memory stats reported ?

sahjhan, 2017-09-25:
Windows 10 both Chrome and Firefox. Tried it in incognito mode too. model is Model Number:ESP-12E with "amica" printed on the silkscreen. 

I get the following which does not change on refresh:
totalBytes:2949250 usedBytes:128261 blockSize:8192 pageSize:256 heap:7912

I am on same LAN / Subnet as the device. I do have another esp8266 on the way but I would have thought none of the pages would have loaded if the device was faulty?

Also worth noting I do not currently have anything connected to it as I wanted to make sure the interface came up ok.

Thanks

allthepies, 2017-09-25:
OK, you're on Chrome, that's good.  On the main BPL (empty) page, go into the developer console Shit-Ctrl-i  select the Console tab and  refresh the main BPL page.  Do any errors appear ?

sahjhan, 2017-09-25:
the console log remains blank

allthepies, 2017-09-25:
OK, that's good.  Have you cleared your browser cache ?

sahjhan, 2017-09-25:
right this is interesting, I have just flashed an old Wemos D1 thats for another project and I get the same issue with that so I do not think its a hardware issue. I am using BrewPiLess.v2.3.3.bin from the bin folder and flashing with NodeMCU-PyFlasher-2.0.exe. Though if it was a flashing issue I would expect no pages to load. It should definitely load without any sensors connected? I have cleared my cache.

allthepies, 2017-09-25:
Hmmm, one for Vito them.   I always build BPL from code using PlatformIO, if you can do that then that would be my next port of call unless Vito can suggest something.

sahjhan, 2017-09-25:
Managed to compile with platformsIO (had not used it before) did a flash erase and write....get the exact same problem..how very odd :(

sahjhan, 2017-09-25:
Ok, interesting development. Got it working in SoftAP mode. I am guessing this is something to do with the wireless AP, but its odd that some of the pages load?

sahjhan, 2017-09-25:
ok, seems I have got it working, connected from another machine and its loading fine now! Ill run wireshark on my pc to see if I can see what is happening with the traffic but we can assume this is not an issue with BrePiLess. At least I learnt more about it in the process. Thanks for the help allthepies!

closed 2017-09-25 by sahjhan

allthepies, 2017-09-25:
No probs, good to hear you've got a way forward.