vitotai/BrewPiLess #36: Fridge cycling on and off frequently (every 10 s or so)

Kollensperger, 2017-09-28:
I am using brewpiless v 2.3.3 on a NodeMCU with OLED in beer mode (symptoms are the same in fridge mode). I am actuating SSRs rather than EM relays although i dont think that makes a difference.
The compressor only comes on for a very short time - 10 s or less, before a short idle period and then comes on again. As far as I understand this is not ideal for the compressor and brewpiless has some code in place for this. I did not enable any glycol settings as far as I am aware.

I also tried modifying the time on http://brewpiless.local/testcmd.htm by sending `j{"minCoolTime":180,"minCoolIdleTime":300}`

The actual cooling time shown on the display is much longer, but the PWM period is short.

Any ideas as to how I can fix this?

allthepies, 2017-09-28:
Hmmm, the default BPL settings have protection delays built in to protect the heater/chiller from such frequent cycling.  Does the BPL OLED or web page show the state being changed from Idling to Cooling at a period which matches the compressor being turned on/off ?  

I've never experienced this issue and can see the compressor being protected  from regular cycling.  I would suggest resetting all parameters to defaults / erasing the EPROM settings etc.

Kollensperger, 2017-09-28:
Thanks for the swift reply. I have reset the EEPROM to no effect - I am now giving it a try with v 2.3.2 uploaded via the web interface (previously I used PlatformIO to program the nodeMCU).

Both the OLED and the webpage show continuous cooling, but the compressor cycles on and off in that period.

allthepies, 2017-09-28:
Ah, I suspect a wiring/hardware issue then, not a BPL s/w issue.

vitotai, 2017-09-29:
Changing minCoolTime/minCoolIdleTime is only available in GLYCOL build. Please verify the settings by issuing the command "c" (single lower case c, stands for control) in testcmd.htm.

It's likely a hardware issue if the screen and web both show cooling.

Makarraum, 2017-10-03:
I have the same problem. And in my case there isn't hardware problem.
Another "problem" is the "Waiting for Peak" periods. They are very long and constants and my temperature climbs a lot.

allthepies, 2017-10-03:
Out of interest, how do you know there isn't a hardware problem ?   Are you using SSRs or mechanical relays ?  Have you tested the relay switching independently of the fridge ?

Makarraum, 2017-10-03:
I have tested the mechanical relays at the same hardware board. The Brewpiless page shows the same as the real world. Switch relay on for 5 or 6 sec and switch it off. Goes to "Waiting for cooling" mode.. After goes to "Waiting for peak" for a looonnngg time..
I realy know what's happening.

allthepies, 2017-10-03:
OK, that's a slightly different pattern to that reported by Kollensperger as you are observing the relay operating in sync with the BPL state change.

Have you modified any of the BPL parameters, either in the code or via the testcmd webpage  ?

Makarraum, 2017-10-03:
No. I just installed the compiled version via OTA interface...

vitotai, 2017-10-03:
@Makarraum
Do you use glycol or fridge? Is that COOLING or HEATING cycle? Which binary file did you use?

The default values of
maxCoolTimeForEstimate is 1200 secs, or 20 minutes
maxHeatTimeForEstimate is 600 secs, or 10 minutes

Have you ever changed the configuration? 

I've never had this issue when using fridge, but I found I have to change it in Glycol setup.

Makarraum, 2017-10-04:
I'm using fridge with cooling cycle only, and the BrewPiLess.v2.3.3.bin file.

I've never changed any configuration parameter, but really 20min is a long time. I'll try to change this values.
I don't know why to turn the fridge on for less than 10 seconds, when the beer temp is 1 degrees above the setpoint...

vitotai, 2017-10-04:
20 minutes is the maximum value. Usually it takes less to get the estimated peak.
The basic idea is that the temperature will keep dropping(when cooling) after the COOLING is off. The algorithm of BrewPi will track this peak and predict the value. 

IMO, your problem is that the compressor is not turned ON, but the controller, the software, thinks cooling is ON. The software expects the temperature to drop to some point,(waiting for peak), but it didn't because the cooling is not ON.

Without GLYCOL option, the minimum cooling time is HARD CODED to 180 seconds. No way you can change it to 10 seconds. This is the real problem.

Makarraum, 2017-10-04:
Yes, I understand how the algorithm works. I see the compressor working and the temperature falling. 
About the rapid cooling time, is not every time it occurs. But it will not to occurs never at this time (less then 10s).

Sorry for my english...

vitotai, 2017-10-04:
I have no idea what goes wrong. The temperature controlling algorithm is ported from BrewPi. I didn't even understand every bit of it. It just works for most of us.

You might already do that, but I would like to remind you that the PIN type should be set to "Not Inverted" while it is  usually "Inverted" for relay modules.

quimnut, 2017-10-04:
If you are pulling power off 3.3v I'd check that with a meter, being mindful an esp8266 can pull up to 850ma and many lm1117's stop at 800ma.

Makarraum, 2017-10-04:
Yes, I changed the Pin to not inverted. The system works properly and do the control, but sometimes it do this.
I think if can be caused by "trash" in esp memory by successive upgrades?

vitotai, 2017-10-04:
@Makarraum
It's not likely that successive upgrade will generate issues like this. 
I do have other idea. If the temperature sensors disconnect or don't respond, the algorithm will stop cooling/heating.  Maybe you can check the sensor connection or replace the sensors.

Makarraum, 2017-10-04:
Perhaps this can occur in a short period of time, such as in one of sensor readings due to noise on the line communication with the sensor. I will try to install decoupling capacitors in the power supply line of the sensors.

Thanks

vitotai, 2017-10-04:
@Makarraum 
I have a 3meter waterproof sensor from eBay. 
It seems to have issue from time to time, especially when used with other sensors. ( Connected together on the same base.) The period is so short that I couldn't see it on the LCD display. How did I know? I was using it in my other project, BrewManiacEX. BrewManiacEx is a brew controller, and it will alarm when the sensors disconnect or don't respond. I had heard the alarm for 4 or more times, but I didn't see the "invalid" temperature when I went to check the LCD.

If it is this case, BrewPi(less) should wait for minimum idle time before turning on the cooling again.

Makarraum, 2017-10-04:
Yes I know. I'll see this with an oscilloscope and install filter capacitor.

Kollensperger, 2017-10-12:
As @allthepies thought, in my case it was a hardware issue. The voltage high signal was not enough to switch the SSRs reliably. After upping the voltage to 5V with a mosfet the problem has disappeared. Thank you all for your input

closed 2017-10-12 by Kollensperger

universam1, 2017-10-12:
A trick in this regard is not to pull the SSR up with the ESP, but to pull it down (invert) and connecting the anode of the SSR to +5V. Hence you don't need a Mosfet circuit. It stays inside the specs of the ESP, I have set it up like this and it works great since almost years ( on a brewing system )

vitotai, 2017-10-13:
@universam1 
I have very little hardware knowledge, and it's like a generic FAQ, not specific to BPL.

Kollensperger, 2018-09-25:
@universam1: I now tried your suggestion as my mosfet failed and it works perfectly. Thank you for your input - I know its a year late but better late than never :). 
 For anyone else considering this:
Connect the anode on the SSR (marked plus) permanently to plus 5V and leave the actuator pin pulled high to 3.3V (by selecting inverted pin in the brewpiless device setup menu) which creates a potential difference too small to switch the SSR. When the brewpiless switches the cooling circuit on, it pulls the relevant pin low, giving a 5V difference which switches the relay more reliably that the 3.3V I could achieve otherwisen.