vitotai/BrewPiLess #39: Sonoff: trading SPIFFS for OTA

universam1, 2017-10-05:
Wondering if it would make more sense to drop SPIFFS and thus logging for enabling OTA.
Alternatively, it could be a choice and available via PIO env.

The layout to enable OTA with 1M flash is outlined nicely here:
https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20170619---configuration-settings-save-locations-in-flash

To enable it simply use this directive:
`build_flags = -Wl,-Tesp8266.flash.1m0.ld`

I think a define `SONOFF_OTA_NO_SPIFFS` could be the switch.

allthepies, 2017-10-05:
OTA is already available/working in BPL, it's how I perform my updates.   Are you talking about BPL running on an ESP which has limited flash memory ?

universam1, 2017-10-05:
Talking about Sonoff stock devices (Flash not replaced)
See here: https://github.com/vitotai/BrewPiLess/blob/master/bins/README.md#sonoff512spiffs

vitotai, 2017-10-05:
@universam1 

I have built that configuration, which is my first attempt because I started this project with only cloud based logging.
SPIFFS is necessary for configuration files, like Beer Profiles.( It should be better to use SPIFFS as storage for BrewPi settings, which should provide wear-leveling. I just don't fell urgency.)

universam1, 2017-10-11:
Just realized that you actually implemented it https://github.com/vitotai/BrewPiLess/blob/master/platformio.ini#L68
Awesome work, I'm amazed, will try!

vitotai, 2017-10-12:
I did it after previous response. I just forget to notice you because of the delay of finding a ESP-01 for verification and being distracted by a new feature which you might be interested.

universam1, 2017-10-12:
I tested the sonoffOTA version on a Sonoff Basic with only DS Sensor added but otherwise stock. Works all fine, so far no issue. I have to check where the constraints are without Spiffs, but so far looks like a platform to recommend. 

Yes I definitely like to hear that! Sounds like a very convenient feature!
BTW Im working on adding a remote configuration option to the iSpindel, that might come handy with the BPL as well.  

One question, is it possible to use the iSpindel Temperature sensor as a source for BPL?

vitotai, 2017-10-13:
@universam1 
I already traced the code and know how to modify the code. However, there are still things that are unclear to me, including
- fallback when no connection for certain time
- filters parameter. My knowledge about signal processing isn't good enough.
- (other issues that might be known until it runs.)

universam1, 2017-10-13:
Maybe I can help a bit in this regard. However, a few ideologies of BPL are not clear to me. When in beer profile mode, how do the two sensors (beer, fridge) interact or play a role? I think this is the critical part how to properly position the iSpindel sensor data.

vitotai, 2017-10-13:
The basic idea of BrewPi is controlling the beer temperature by controlling the fridge temperature.

There is a Fridge Constant mode, which is simply a dead-zone algorithm. 

The Beer Profile mode is a special Beer Constant mode, in which the beer set point is set by the host. The host regularly checks the BeerSet and changes it according to the profile. In original BrewPi, a RaspberryPi is used to adjust the temperature. In BPL, it is done by ESP8266 directly.

PID algorithm is used to determine the Set point of fridge temperature(FridgeSet). BeerTemp is the input of the PID, and FridgeSet is the output. Based on the calculated FridgeSet and current FridgeTemp, the algorithm determine whether cooling or heating is necessary. 
eg.
If FridgeTemp > (FridgeSet + Offset), start cooling until Fridge < (FrdigeSet +Offset + estimated peak.) 
Offsets are settable options. The estimated peak is an accumulated parameter. You might find that sometimes the LCD shows "waiting for peak". That is when BrewPi algorithm trying to get the estimated peak.

BrewPi was designed to be used with fridge as fermenting chamber. The algorithm assumes two sensors. Therefore, to use it in Glycol setup, FallBackSensor is used to trick the algorithm.

vitotai, 2017-10-17:
I did the implementation. I just verified on my test platform.
https://github.com/vitotai/BrewPiLess/tree/v24-ispindel#using-ispindel-as-beer-temperature-sensor

universam1, 2017-10-17:
Awesome - I will try asap! Great work!