vitotai/BrewPiLess #41: SettableMinimumCoolTime without EnableGlycolSupport

universam1, 2017-10-13:
The hardcoded defines in https://github.com/vitotai/BrewPiLess/blob/3dee90c54666e5b706de558b6f982ab0839380dd/src/TempControl.h#L34 are too long for me, but I don't have a glycol setup. 
Trying to understand, why it is prevented to tune this parameter outside of `EnableGlycolSupport` mode.

Any thoughts?

vitotai, 2017-10-13:
The Glycol option includes two features.
SettableMinimumCoolTime and
FridgeSensorFallBack.

FridgeSensorFallBack is to use Beer Sensor for fridge temperature reading if fridge sensor is not present. If Fridge sensor is assigned and working, it works just what it works without this option.

Most people use BrewPi(or BrewPiLess) for fridge, and wrong settings might damage the compressor. That's why this option is not default ON. You can change it in Config.h. The configuration file will be different, and you will need to run setup again.

universam1, 2017-10-14:
Thanks for the explanation!

It's of course your call how you implement features, but given that you set already sane presets, double hurdles to be able to edit this defaults are not necessary. If a user wants to change that setting, he will anyway... Just my 2c

universam1, 2017-10-15:
@vitotai Readme says:

`build with GlycolSupport option enabled It's simple by adding -DGlycolSupport=true to build flag in platformio.ini

build_flags = -Wl,-Tesp8266.flash.4m.ld -DGlycolSupport=true`

There is no define `GlycolSupport` - are you meaning `EnableGlycolSupport`?

vitotai, 2017-10-17:
You've made a good point.
I was just too conservative.