vitotai/BrewPiLess #42: Sonoff needs flash mode DOUT

universam1, 2017-10-15:
Faced the problem that my Sonoff does not start reliable, even being dead completelly.

Found this and solved my issues:
https://github.com/arendst/Sonoff-Tasmota/blob/66d9ff14ac804ea6f63fe6e2f371357a138d3bba/README.md#attention-all-versions

Therefore, platformio.ini needs these settings:

`board_flash_mode = dout`

vitotai, 2017-10-17:
Thanks.

I googled and found that it is for the newer SONOFF. The old version seems to work find without that options.

universam1, 2017-10-17:
Possible. My recently purchased Sonoff Basic is repeatable unable to start without this flash mode. That's a bit weird as the documentation would rather mention the ESP8285, but the test against different firmwares proves this true. It might be a causality, but it does not hurt anyway.