vitotai/BrewPiLess #43: Error compiling v2.3.3 in plataform IO (athom) w/Rotary Encoder

diegovazquez, 2017-11-01:
Hi, 

I have changed the following in config.h = #define BREWPI_ROTARY_ENCODER 1

Everytime i try to build the the project I get the following result:

> 
> 18.7 s
> platformio run --target upload
> Compiling .pioenvs\nodemcuv2\src\RotaryEncoder.o
> Compiling .pioenvs\nodemcuv2\src\Sensor.o
> Compiling .pioenvs\nodemcuv2\src\SettingsManager.o
> Compiling .pioenvs\nodemcuv2\src\Simulator.o
> Compiling .pioenvs\nodemcuv2\src\SpiLcd.o
> src\RotaryEncoder.cpp:199:1: error: 'PCF8574' does not name a type
> PCF8574 pcf8574(PCF8574_ADDRESS,PIN_SDA, PIN_SCL);
> 
> ^
> src\RotaryEncoder.cpp: In static member function 'static void RotaryEncoder::process()':
> src\RotaryEncoder.cpp:259:12: error: 'pcf8574' was not declared in this scope
> uint8_t p=pcf8574.read8();
> ^
> *** [.pioenvs\nodemcuv2\src\RotaryEncoder.o] Error 1
>  [ERROR] Took 16.69 seconds
>  
>  [SUMMARY]
> Environment glycol      [SKIP]
> Environment oled        [SKIP]
> Environment thorrax     [SKIP]
> Environment sonoff      [SKIP]
> Environment sonoffota   [SKIP]
> Environment nodemcuv2   [ERROR]
>  [ERROR] Took 16.69 seconds
>  
> Linter
> Severity 	Provider 	Description	Line 
> Warning	GCC	#pragma once in main file [enabled by default]	21:9

thank you very much

vitotai, 2017-11-01:
Please edit the platform.ini. around line 20
remove the line of "PCF8574_ESP".

I don't understand the library search rules well. sorry about that.

diegovazquez, 2017-11-01:
It works! thank you

closed 2017-11-01 by diegovazquez