vitotai/BrewPiLess #50: Waiting to heat???

arcidodo, 2017-11-16:
hi there, why does te system want to wait 14 housr?

[image]

vitotai, 2017-11-16:
Did you change the following settings?
minHeatTime
minHeatIdleTime
mutexDeadTime

RaceMouse-of-Denmark, 2017-12-10:
Hi,
I Added this to `void TempControl::init(void)`
```
  DBG_PRINTF("tempFormat=%i\r\n",cc.tempFormat);

  DBG_PRINTF("tempSettingMax=%i\r\n",    intToTempDiff(cc.tempSettingMax)    );
  DBG_PRINTF("tempSettingMin=%i\r\n",    intToTempDiff(cc.tempSettingMin)    );
  DBG_PRINTF("Kd=%i\r\n",                intToTempDiff(cc.Kd)                );
  DBG_PRINTF("Ki=%i\r\n",                intToTempDiff(cc.Ki)                );
  DBG_PRINTF("Kp=%i\r\n",                intToTempDiff(cc.Kp)                );
  DBG_PRINTF("iMaxError=%i\r\n",         intToTempDiff(cc.iMaxError)         );
  DBG_PRINTF("idleRangeHigh=%i\r\n",     intToTempDiff(cc.idleRangeHigh)     );
  DBG_PRINTF("idleRangeLow=%i\r\n",      intToTempDiff(cc.idleRangeLow)      );
  DBG_PRINTF("heatingTargetUpper=%i\r\n",intToTempDiff(cc.heatingTargetUpper));
  DBG_PRINTF("heatingTargetLower=%i\r\n",intToTempDiff(cc.heatingTargetLower));
  DBG_PRINTF("coolingTargetUpper=%i\r\n",intToTempDiff(cc.coolingTargetUpper));
  DBG_PRINTF("coolingTargetLower=%i\r\n",intToTempDiff(cc.coolingTargetLower));

  DBG_PRINTF("maxHeatTimeForEstimate=%u\r\n",cc.maxHeatTimeForEstimate);
  DBG_PRINTF("maxCoolTimeForEstimate=%u\r\n",cc.maxCoolTimeForEstimate);

  DBG_PRINTF("fridgeFastFilter=%u\r\n",cc.fridgeFastFilter);
  DBG_PRINTF("fridgeSlopeFilter=%u\r\n",cc.fridgeSlopeFilter);
  DBG_PRINTF("fridgeSlowFilter=%u\r\n",cc.fridgeSlowFilter);
  DBG_PRINTF("beerFastFilter=%u\r\n",cc.beerFastFilter);
  DBG_PRINTF("beerSlopeFilter=%u\r\n",cc.beerSlopeFilter);
  DBG_PRINTF("beerSlowFilter=%u\r\n",cc.beerSlowFilter);
  DBG_PRINTF("lightAsHeater=%u\r\n",cc.lightAsHeater);
  DBG_PRINTF("rotaryHalfSteps=%u\r\n",cc.rotaryHalfSteps);
  DBG_PRINTF("pidMax=%i\r\n",intToTempDiff(cc.pidMax));

  DBG_PRINTF("minCoolTime=%u\r\n",cc.minCoolTime);
  DBG_PRINTF("minCoolIdleTime=%u\r\n",cc.minCoolIdleTime);
  DBG_PRINTF("minHeatTime=%u\r\n",cc.minHeatTime);
  DBG_PRINTF("minHeatIdleTime=%u\r\n",cc.minHeatIdleTime);
  DBG_PRINTF("mutexDeadTime=%u\r\n",cc.mutexDeadTime);
```

The debug terminal (serial) then printed this at startup:
```
tempSettingMax=0
tempSettingMin=0
Kd=0
Ki=0
Kp=0
iMaxError=0
idleRangeHigh=0
idleRangeLow=0
heatingTargetUpper=0
heatingTargetLower=0
coolingTargetUpper=0
coolingTargetLower=0
maxHeatTimeForEstimate=0
maxCoolTimeForEstimate=0
fridgeFastFilter=0
fridgeSlopeFilter=0
fridgeSlowFilter=0
beerFastFilter=0
beerSlopeFilter=0
beerSlowFilter=0
lightAsHeater=0
rotaryHalfSteps=0
pidMax=0
minCoolTime=0
minCoolIdleTime=0
minHeatTime=0
minHeatIdleTime=0
mutexDeadTime=0

```
After performing a "Erase EEProm" from the setup page the debug printed this:
```
tempSettingMax=-4718592
tempSettingMin=-12320768
Kd=-393216
Ki=65536
Kp=1310720
iMaxError=131072
idleRangeHigh=262144
idleRangeLow=-262144
heatingTargetUpper=78336
heatingTargetLower=-52224
coolingTargetUpper=52224
coolingTargetLower=-78336
maxHeatTimeForEstimate=600
maxCoolTimeForEstimate=1200
fridgeFastFilter=1
fridgeSlopeFilter=3
fridgeSlowFilter=4
beerFastFilter=3
beerSlopeFilter=4
beerSlowFilter=4
lightAsHeater=0
rotaryHalfSteps=0
pidMax=2621440
minCoolTime=180
minCoolIdleTime=300
minHeatTime=180
minHeatIdleTime=300
mutexDeadTime=600

```
This might be an issue. I ran into this one as well.

Shouldn't we also have an interface that can adjust these settings?

//The Race

allthepies, 2017-12-10:
You can manipulate them via:  https://github.com/vitotai/BrewPiLess#json-commands

quimnut, 2017-12-10:
Witnessed this behavior if the eeprom isn't erased after upgrading.
eg. "minHeatTime":32768,"minHeatIdleTime":32768, 
after erase; "minHeatTime":180,"minHeatIdleTime":300,

Geffi81, 2018-02-06:
Where can i find the settings?

vitotai, 2018-02-07:
the page is moved to wiki:
https://github.com/vitotai/BrewPiLess/wiki/JSON-command