vitotai/BrewPiLess #51: BrewPiLess web stop on "Communicating with BrewPiLess controller"

diegovazquez, 2017-11-22:
hi! i dont have internet in site, and BrewPiLess web won't start.

[image]

Thank You

vitotai, 2017-11-23:
Obviously, you are running BPL in AP mode, aren't you?

The charting library not included. Please read the README. There is a section for it.

You can alternatively use Tom's front-end, which embeds the charting library.
Check:
https://github.com/vitotai/BrewPiLess/issues/21

BTW, did you change the index.htm?
The original index.htm should be HTTP only, like
<script src="http://cdnjs.cloudflare.com/ajax/libs/dygraph/1.1.1/dygraph-combined.js"></script>

diegovazquez, 2017-11-23:
Yes, i'm running in AP mode, I upload the lib (see photo) and get the same problem.

[image]

thank you for your help, sorry for my english, is not so good.

Excellent project.

vitotai, 2017-11-23:
Your index.htm requests https, which should be http. So uploading that lib doesn't work.

You can manually "cache" the library. Connect to internet, and  click the the following url. After your browser got this file, reconnect to BrewPiLess.
https://cdnjs.cloudflare.com/ajax/libs/dygraph/1.1.1/dygraph-combined.js

Alternatively, use Tom's front-end as I have said.

closed 2017-11-29 by diegovazquez

RollyBrewer, 2018-07-12:
Strangely, i get this same "Communicating with controller" error on v3rc6, which i believe uses the new front end BUT the error only happens only on my windows laptop. On my MacBook everything is fine, all laptops and BPL are on the same wifi and BPL is in station mode.

Is there some security setting or something per browser maybe?

vitotai, 2018-07-13:
IE and Edge are not supported.

dotps1, 2018-07-14:
is there any plans to support IE or Edge?  I am working on setting this up, I hate the idea of needing a secondary browser just to use this webpage.

vitotai, 2018-07-15:
IE and Edge don't support SSE, or Server Side Event, which is used by BrewPiLess.
WebSocket is a solution. However, the version of AsyncTCP/AsyncWebServer used is not very stable. So, I was forced to sticked with SSE. The latest version seems to be OK, although still not as stable as SSE. Unfortunately, the image will exceed the limit to run on-line update if latest version is used. There are ways. However, I bet a lot of how and why will appear, so I don't jump on them.

vitotai, 2018-07-20:
If you haven't found it, it is here:
https://github.com/vitotai/BrewPiLess/tree/master/htmljs/dist

dotps1, 2018-12-18:
Just an FYI, edge now supports SSE it seems

[image]

Im using edge to configure and monitor now.