vitotai/BrewPiLess #54: Calculated by BPL gone

j0hnsmith, 2017-11-28:
In `2.4` the 'Calculated by BPL' option on the gravity config page has gone (btw, still in README.md), so even though I've added the coefficients this request doesn't result in a gravity reading.

```
curl http://brewpiless/gravity -d '{"name":"iSpindel000","ID":3965675,"angle":77.76,"temperature":26.0625,"battery":4.723671,"gravity":0}' -v -H 'Content-Type: application/json'
```
I get a 200 and this timestamp gets updated.
[image]
Also, iSpindel temperature is showing on the chart, but not gravity :cry:

Here's the config I have 
[image]

Not sure if this is this expected?

vitotai, 2018-01-16:
I just fixed it. Please try latest v2.4.2x

j0hnsmith, 2018-02-20:
@vitotai I'm finally preparing for my next brew but there's no `2.4.2` tag, which commit should I use?

j0hnsmith, 2018-02-20:
nevermind, found it.

j0hnsmith, 2018-02-22:
Works, thanks.

closed 2018-02-22 by j0hnsmith