vitotai/BrewPiLess #55: A lot of unexpected restarts

rflemming, 2018-01-05:
I just finished building an ispindel so I was testing out integration with brewpiless (using SoftAP mode) and I've been seeing a lot of random restarts. I'm assuming they are caused by the watchdog timer kicking in for some reason or another.  I don't believe they are related to ispindel specifically since I'm pretty sure they happened in the past, just maybe not with the same frequency.  You can see all the resume events on the chart:

[image]

Is there anything I can do to help debug this further?  I tried keeping a window open with the log from testcmd.htm but I haven't been able to capture anything that appears useful.  I'm happy to help I just don't know what sort of info would be beneficial.

vitotai, 2018-01-06:
I don't know either.
The only way I would suggest is to check your power supply and ESP8266.

A friend on HBT had the same issue, and he ended up replacing his NodeMcu board to get good result.

It's funny that I was thinking about exception handling of ESP8266. Unfortunately, the source code is not accessible, and we have nothing we can do.

quimnut, 2018-01-06:
esp8266's have been said to use up to 850mA of 3.3v which is also about what lm1117's are rated to, you can try moving the relay power off the 3.3v pin if you do that.

quimnut, 2018-01-14:
On the topic of restarts, I upgraded to 2.4.2 and connected an iSpindel yesterday.

[image]

rflemming, 2018-01-15:
Yup, pretty much what I was seeing.  Did you by chance update to 2.4.2 via the file upload or did you do it over USB?  I did originally did it via the OTA update, but yesterday I decided to pull it out, compile my own 2.4.2 and reflash it via USB.  So far I don't see any resume events in the UI, but I'm also not willing to say they aren't happening anymore yet either.

The problem I'm seeing now (though evidently it's been happening for almost 3 weeks) is that updates aren't showing up at Ubidots.  Nothing changed on my side, so I'm not sure what's going on and I haven't spent a lot of time trying to debug it yet.

quimnut, 2018-01-15:
Yes I did, I just put a 100uF cap on the 3v3 pin to see if it was noise/power but it's no different. It's taking 3 attempts to just get the web interface to load.

Was my first go at using the ispindel yesterday but I noticed if I didn't reset then go in via softap and save the settings it wouldn't start sending data. If I just turned it on after saving things previously it wouldn't do anything obvious, With the softap/save method i see it renew it's dhcp lease every upload interval.

quimnut, 2018-01-15:
usb flashing has fixed the stability for me too. Now to figure out the rest :)

rflemming, 2018-01-16:
Interesting that flashing via USB seems to have helped you as well.  I'm not seeing the resets anymore, but the web UI does still seem to hang every now and then.  As for my issue with Ubidots, that's working again as well.  I suspect disabling and re-enabling may have done the trick.

BB-grooves, 2018-01-19:
Are you using a OLED display?

with latest firmware there is not a lot of heapspace - when it gets low you will experience no responses on the webserver and restarts as the watchdog reboots the ESP8266.

For OLED add -DOLEDDISPLAY_REDUCE_MEMORY=true to your build

heap space is low even without the 8k / 16k for buffering the OLED. So try reducing the various buffer sizes from 2048 & 1024 to 512 - however this might cause other issues for large configurations/profiles. I'm currently trying to find optimal values.

rflemming, 2018-01-19:
Nope, LCD for me.

vitotai, 2018-01-19:
FYI, you can check the heap status by this URL, like
http://brewpiless.local/fs

I would suggest to get another D1 mini or NodeMcu. They are cheap and people have bad copies from time to time. (Luckily, I don't.)

roguenorman, 2019-02-04:
my setup was working for a week and then all of a sudden it started to restart every 1min 30 seconds-ish. I am using a D1 mini and also tried a spare that I have but I was getting the same results. I was on version 3.4. I checked the heap size and it is around 16000. I ended up disabling the LCD, buttons and the buzzer but I'm not sure if that would lower heap size. I used 3.5 and finally got it to stay on for 10 minutes so my cooling cycle would kick in. is 16000 heap a lot? I had it set to fridge const and it seemed stable but when i set it to beer const that is when the issues started. Should that matter? or is it coincidence. Is it even the heap making it reset?? I would not even care that much about restarts but it makes the timer start again so the minCoolIdleTime gets reset everytime so i pretty much never get a cooling cycle

quimnut, 2019-02-04:
I would try formatting spiffs if you have not already.

roguenorman, 2019-02-04:
Yeah actually I did do that just before I got it working again. I lost track of what changes I was making after a while but that could definitely be the thing that fixed it.