vitotai/BrewPiLess #57: Brew and calibrate not working?

rflemming, 2018-01-16:
I can't seem to get ispindel/BPL to work in Brew and Calibrate mode.  The iSpindel is talking to BPL just fine and I can see data coming in:

16:7:33 ↓ G:{"name":"iSpindel001","ID":3514868,"angle":55.30649,"temperature":18.3125,"battery":5.338894,"gravity":15.80349}
16:7:33 ↓ G:{"name":"iSpindel001","battery":5.34,"sg":15.803,"angle":0.00,"lu":1516061252,"lpf":0.10,"stpt":1}
16:9:31 ↓ G:{"name":"iSpindel001","ID":3514868,"angle":55.31021,"temperature":18.3125,"battery":5.338894,"gravity":15.80543}
16:9:31 ↓ G:{"name":"iSpindel001","battery":5.34,"sg":15.805,"angle":0.00,"lu":1516061371,"lpf":0.10,"stpt":1}

When setting up the gravity sensor I selected iSpindel and Calibrate iSpindel and set the tilt in water prior to turning on logging.  Then I enabled local logging and entered the OG and SG which was 1.060 when I started.  I do not see the symbol in the lower right corner of the legend indicating Brew and Calibrate mode.  In addition I see the SG value from the iSpindel being reported in the UI in Plato, and the tilt value is always 0:

[image]

Even prior to trying to do Brew and Calibrate the tilt value was never being reported despite it being in the received.  Any help is appreciated.

vitotai, 2018-01-16:
Read it carefully:
https://github.com/vitotai/BrewPiLess/wiki/Brew-and-Calibrate-iSpindel

I think you at least skip step 5.

rflemming, 2018-01-16:
If you read my original issue I did those things:

[image]

Then setup local logging:

[image]

and recorded OG/SG as 1.060, which I can't screenshot now since the SG is being altered with the value from the iSpindel.

Then I put in the OG/SG.  I've stopped logging and repeated these steps numerous times and still no luck.  Also note that the tilt value is not being reported in the UI despite it being reported by the iSpindel:

20:12:25 ↓ G:{"name":"iSpindel001","ID":3514868,"angle":54.89464,"temperature":18.8125,"battery":5.338894,"gravity":15.58815}

20:12:25 ↓ G:{"name":"iSpindel001","battery":5.34,"sg":15.588,"angle":0.00,"lu":1516075944,"lpf":0.10,"stpt":1}

Note the angle in the first message which I'm assuming comes from the iSpindel iteself, but the angle is absent from the second message which I'm assuming is BPL related.

vitotai, 2018-01-16:
apologize. I didn't read the text carefully.

That issue should be fixed a few weeks ago. Let me check.

rflemming, 2018-01-16:
I built the version of 2.4.2 that I'm running from the master branch a few days ago.

vitotai, 2018-01-16:
Well.
I did NOT fix that bug. Sorry about that.
I just pushed a version 2.4.2x. (Because I am in the progress of v2.5, it takes time for me to change the version number.)

rflemming, 2018-01-16:
In addition to the 0 tilt issue does that resolve the brew and calibrate problem?

vitotai, 2018-01-16:
I am not sure about the problem you have.
I brewed with that twice.

rflemming, 2018-02-27:
I haven't tried it, but I suspect that the Brew and Calibrate issue will be fixed by the solution here: https://github.com/vitotai/BrewPiLess/issues/68

closed 2018-02-27 by rflemming