vitotai/BrewPiLess #58: BPL loses static IP settings on power off

lekrom, 2018-01-17:
Hi Vito,

My BPL setup is working pretty great with several fermentation successfully completed.
I setup my BPL with a static IP and set the default gateway to my router IP. (For the purpose of my iSpindel)
I have however picked up a slight issue. We had a few power failures over the last few days. 
When the system comes back online after power is restored, it gets a dynamic IP allocation, seems to lose the static IP and gateway settings somehow. I erased the WiFi settings and reset it. It worked fine until the next power failure. So far I have not noticed any other settings going missing.

Regards,
Lekrom

vitotai, 2018-01-17:
I just revised WiFi Configuration part. Please try v2.5.

lekrom, 2018-01-17:
Thanks Vito. I will try it in a few days. In the middle of an experimental saisson. Will change FW when its done.

vitotai, 2018-01-17:
I think I found the bug. Just fixed.