vitotai/BrewPiLess #59: New version breaks BPL

allthepies, 2018-01-17:
Hi Vito,

Just tried the new version and it doesn't work for me.

The latest libs version  just goes  into a reboot loop, the legacy libs version "works" but the LCD web screen can't connect to the controller.  Have cleared browser cache multiple times.

allthepies, 2018-01-17:
OK, tracked down the LCD update issue.

I had an existing logging session going on when I upgraded and that seems to have caused an infinite loop in the resumeSession() method in Brewlogger.h.

It was spinning round the "while(totalRead < fsize){" loop constantly writing out :

Move 2 bytes, read:0 process:0

So I wiped SPIFFS and attempted again.  Both the new libs and legacy libs variants of 2.5 now work but with the new async libs version the Device Setup page wouldn't load, the browser never got a response.  All other pages were OK.

Reverted back the the legacy libs variant and all pages working OK.

vitotai, 2018-01-17:
I am sorry.
v2.5 is not really ready. Especially for the latest libraries with latest framework.

The latest Async libraries seems to have issue with 1.5 framework. The README.md of ESPAsyncWebServer states that latest framework "might" be necessary. I've tested different combinations for a while but couldn't draw a conclusion.

One big issue about latest framework is that it is 25.5k bigger than release 1.5. BPL is almost near the boundary of 512k. The image size built with latest 1.6 framework is too big, which might break the flash layout and result in unpredictable condition. In the best case, you will lost OTA function after that. This is exact the reason why I keep using the legacy version. 

There are some fixes and some revision so I think a release before migrating to latest framework is necessary.

vitotai, 2018-01-18:
Some information for you:
http://vito.tw/?p=895

vitotai, 2018-01-18:
I will look into it. 

BTW, I am trying 2M sketch space with 2M SPIFFS configuration. It works fine, but I found one thing weird. After I flashed the 2M version onto the originally running 1M sketch space version, I expected that the SPIFFS will be created at different address on the flash. However, there were files left. so, SPIFFS might not work as what I have thought.

allthepies, 2018-01-19:
A new version Vito!   platformio file mentions a new flash script ?   What do I need to do in order to build/flash the latest version ?

vitotai, 2018-01-19:
Check: https://github.com/vitotai/BrewPiLess/wiki/2M-sketch-Space

allthepies, 2018-01-19:
Thanks,

I'll look into creating a custom framework for BPL as I don't want that config applying globally.

allthepies, 2018-01-19:
Thanks!

v2.5 now running with latest libs :)

I'm using an OLED and have added the reduce memory directive in my build, seems to slow the OLED refresh down slightly but no crazy flickering or any other issues.  I wonder if there is a text-only OLED library ?  Libraries are great but they often contain lots of baggage which just might not be required for a specific application.

vitotai, 2018-01-19:
The OLED is basically graphics based. It seems to me that font is necessary since there is no font rom built-in.
During linking phase,  the compiler(linker, to be precise) will eliminate unused functions and global variables. We don't need to worry about that.