vitotai/BrewPiLess #60: http authentication change

allthepies, 2018-02-06:
The http authentication method seems to have changed recently (for the pages which require authentication) from the "Basic" mechanism to a "Digest" mechanism.   I imagine this is down to the newer AsyncWebServer library but I couldn't find the actual change in the library which would have switched the mechanism.

BPL doesn't specify the authorisation type to use when it invokes the requestAuthentication() method and so it looks like it will default to using a digest :-

 void requestAuthentication(const char * realm = NULL, bool isDigest = true);

But this is the same signature as for the legacy library and so I don't know why the previous version used basic authentication.

The reason I bring this up is my Alexa skill which can change the BPL mode invokes a BPL webservice and was using basic authentication to pass the BPL user/password.  However this no longer works.

vitotai, 2018-02-07:
Well, I remember that HttpWebServer(synchronized version) supports only basic. I am not aware of types of authentication.
To debug this, sniffer like wireshark is necessary. However, I am not familiar with wireshark. 

BTW, I found the latest version isn't as good as old one. My browser occasionally takes forever to fetch the pages while I can get the html by using CURL at the same time. Again, sniffer is necessary...

allthepies, 2018-02-07:
I don't know about the need to use wireshark, you can see the http authentication mechanism being requested/used in the http headers using Chrome dev tools.
I agree that the new libs do give us challenges in BPL, I find that with logging on then when the log data file gets above a certain limit then the GET /chart.php request to fetch all of the chart data never completes successfully.  It gets timed out after 5 seconds, which is the value specified in the BPL javascript, Ideally I'd like that timeout extended.  (My BPL is running in a garage attached to my house where the Wifi router is and I only get a ~23 Wifi signal reported by BPL so I guess there's a lot of error correction going on and this slows down the BPL connection).

allthepies, 2018-02-10:
OK, the authentication issue is a red herring.  Closing

closed 2018-02-10 by allthepies