vitotai/BrewPiLess #62: Brewpiless on wemos triple board

riisefrutti, 2018-02-16:
Hi,

Sorry, this is not an issue - more of a curiosity. Have you tried to use this on the wemos triple(with two relays and one wemos)? Would be a nice space saver - in my setup that is.

Been running BPL a year now flawlessly :D Gep up the great work.

vitotai, 2018-02-18:
you will have to make sure that the control pin of the relay module is configurable.

jamesgol, 2018-02-21:
I've used multiple relays like those before, unfortunately the GPIO is not selectable so you need to cut the trace on the board and solder a jumper to the new one.

riisefrutti, 2018-03-14:
Got it working, not a recommended setup. 
What buttons do you recommend @vitotai for brewpiless? Could you share a link?

vitotai, 2018-03-14:
I happen to have two buttons left and used them. I don't recommend them, though. The reason I bought them is that they are round, small, easy to mount, and water-resistant. However, there is no feedback on press.

riisefrutti, 2018-03-14:
Do you have any recommended buttons?

vitotai, 2018-03-15:
I have not find a good one.