vitotai/BrewPiLess #63: temp control issue

lekrom, 2018-02-17:


lekrom, 2018-02-17:
Hi Vito.
I recently upgraded to ver 2.5.1. I have a strange issue. The temperature control no longer works. The cooling cycle starts but never stops. I have a fridge full of frozen beer. The temp was set to 20 degrees (fridge constant). Something obviously went very wrong.  I switched ot off and tried again. Same result.

Lekrom.

vitotai, 2018-02-18:
Upgrading from v2.3 and before requires re-initialization of EEPROM, which is simply by "erasing" EEPROM at "Device Setup" page.

To check the temperature control settings, you can use testcmd.htm, and issue "c" command.
https://github.com/vitotai/BrewPiLess/wiki/JSON-command

j0hnsmith, 2018-02-25:
I've got the same, the starsan in my airlock is frozen, beer temp is currently 7.9ºC :( I hope I haven't killed all the yeast. Didn't realise the EEPROM needed to be reset.

j0hnsmith, 2018-02-26:
Works as expected after EEPROM reset.