vitotai/BrewPiLess #64: Rotary encoder causes crashes and restarts

sryburn, 2018-02-20:
I am using BPL with a rotary encoder via PCF8574. While this basically works ok, spinning the encoder too fast will consistently cause NodeMCU to either reboot or crash (fully unresponsive). This only occurs when the menu is active (i.e. changing mode or setting temperature).

Not sure how to further debug this issue, but any suggestions welcome!

vitotai, 2018-02-20:
I don't recommend using rotary encoder with PCF8574. It is implemented just because it can be done, but it is not well tested.

Interrupt is necessary and used for rotary encoders and PCF8574 is accessed via I2C that is used to control by LCD too. The result of rotary encoder processing usually results in update of display, which will easily result in mutual access of I2C. Let me illustrate to make it clear.
(rotate the encoder) -> Interrupt -> read PCF8574 via I2C -> Update LCD (via I2C)
If the encoder is rooted so fast that the update of display is interrupted, the I2C protocol will be broken. To prevent it, interrupt is disabled during updating LCD. (I didn't modify the library for OLED LCD, so OLED LCD won't work with rotary encoder.)
To make things worse, ESP8266 doesn't have hardware I2C, and it is software I2C that we are using.

Potential issues I have thought:
1. slow processing of I2C, resulting in slow Interrupt processing.
2. slow processing of I2C, resulting in long period of blocking of interrupt for display
3. frequent and long period of disabling interrupt might result in system instability.

Verifying and debuting the issues above without a scope or logical analyzer is beyond my ability.

sryburn, 2018-02-21:
Thanks for the info, I will have a play and update if I find a solution.

sryburn, 2018-02-22:
Commenting out the `wait.millis(3);` lines in Menu.cpp resolves the problem. I guess the issue is related to disabling interrupts during the delay. Obviously this prevents the menu from blinking the selected item - perhaps this could be implemented in an alternative way?

vitotai, 2018-02-22:
Awesome!

I didn't study this part before. Based on my initial judgement, the blinking is control by `blinkTimer`. 

The code:
`
if (blinkTimer==0)
			show();
		else if (blinkTimer==128)
			hide();
`

blinkTimer is a uint8, so 128 is the middle. Each cycle is  255 * (3ms + time for each loop).
We can eliminate the delay by checking time, a simple solution will be something like

#define BLINKING_TIME 250
bool blinkLoop(
	void (*changed)(),	// called to update the value
	void (*show)(),		// called to show the current value
	void (*hide)(),	// called to blank out the current value
	void (*pushed)())	// handle selection
{
	uint16_t lastChangeTime = ticks.seconds();
	//uint8_t blinkTimer = 0;
	uint32_t switchTime=0;
	bool hidden=true;

	while(ticks.timeSince(lastChangeTime) < MENU_TIMEOUT){ // time out at 10 seconds
	//#if ESP8266
		yield();
	//#endif
		if(rotaryEncoder.changed()){
			lastChangeTime = ticks.seconds();
			//blinkTimer = 0;
			switchTime=0;
			hidden=true;
			changed();
		}
/*		if (blinkTimer==0)
			show();
		else if (blinkTimer==128)
			hide(); */
		uint32_t current=millis();
		if((current - switchTime) > BLINKING_TIME){
			if(hidden) show();
			else hide();
			hidden = ! hidden;
			switchTime = current;
		}

		if (rotaryEncoder.pushed()) {
			rotaryEncoder.resetPushed();
			show();
			pushed();
			return true;
		}

		//blinkTimer++;
		//wait.millis(3); // delay for blinking
	}
	return false;
}

I don't have access to any hardware until late March. Before then, I can't verify it. You can try, though.

sryburn, 2018-02-23:
Great! I can confirm this works perfectly =D

sryburn, 2018-02-26:
All good, cheers!

closed 2018-02-26 by sryburn