vitotai/BrewPiLess #65: Wrong WIFI password after fresh flash

gh112, 2018-02-20:
My BPL stopped responding after i tried to change from Station to Station + AP. 

I performed a fresh flash with the prebuilt images. 
After the flash i can see the new network ( brewpiless ) however i cannot log in.
I have tried with 2.5 and 2.5.1. I have not seen any change of SSID / Pass in the documentation.

vitotai, 2018-02-21:
https://github.com/vitotai/BrewPiLess/wiki/System-Setup

gh112, 2018-02-21:
Yes i know the SSID/Pass should be brewpiless/brewpiless
It does not work after a flash.  The SSID is brewpiless but the password is not brewpiless

I have read that the ESP8266 stores wifi credentials somewhere else. Could this be the problem?

vitotai, 2018-02-22:
hostname = SSID
network pass phrase = password
are all "brewpiless" by default.

If you can plug the ESP8266 to USB, try to turn on serial debugging option, the debugging log will printout username, ssid, and password.

gh112, 2018-02-24:
Thank you vito!
I have sucessfully connected to the BPL again. 

I also want to thank you for amazing work on this project and integration with iSpindel.

closed 2018-02-26 by gh112