vitotai/BrewPiLess #67: Issue wiith temp sensor

pdrsn, 2018-02-25:
I changed the ESP to a newer one with v2.5 on my setup and after this, I get sudden jumps in value from one temperature sensor. It goes to exactly the same value as another sensor for a while, then jumps back to the real value. Any idea what may cause this?
[image]

vitotai, 2018-02-26:
sensor disconnection. The default(invalid) valid of temperature is 85C. Sometimes, the first readings from the sensors are 85C. The values in the chart are "filtered", therefore you don't see the raw values.

It might be a bug, though.

pdrsn, 2018-03-01:
Here is a closeup of one of the jumps. I will have to check the wireing later
[image]

insnhx, 2018-03-04:
I have the same exact issue. The chamber probe jumps around and reads same exact value of the beer probe most of the time. And for example when it says “waiting to heat” and then the time expires one of the probes jumps in temperature and the heating never happens. I am running the latest version of brewpiless on this module. Btw. I have another module that worked flawlessly with an older version.

I will check all the connections on this module but I think everything is soldered fine, but will definitely report if I find anything. Keep up the good work!!

vstoms, 2018-03-10:
Have the same issue after updating to 2.5.1
[image]

insnhx, 2018-03-14:
The error I was getting was on the amica nodemcu board. I never checked wiring but rewired everything to d1mini, installed the latest brewpiless and can confirm that the problem is gone! 
So I dont really know if the issue was with the board version and that specific brewpiless version, or just the wiring.

vitotai, 2018-03-15:
I would believe it's hardware issue.
First of all, the sensor related code is derived from BrewPi. I've tried my best not to touch that part while I don't have a reason to. This part of code was ported by Thorrax, who did it before I tried to do. That says, Thorrax's native ESP8266 BrewPi shares the same code.
Second, it seems that not everyone has this issue.

FYI, I had two setups. One of them had sensor disconnection issue while the other didn't. I checked and rewired for a few times, but it still happened. I still couldn't solve this issue.