vitotai/BrewPiLess #68: Gravity/Logging Settings appear to be ignored

rflemming, 2018-02-25:
This may go back to: https://github.com/vitotai/BrewPiLess/issues/57  I manually calibrated by iSpindel and put the coefficients (as calculated by calibrationSG.htm) into BPL yet they appear to be being ignored:

[image]

The tilt value is being reported properly, but it appears like the gravity reading from the iSpindel is being used (which is in Plato) rather than BPL doing the calculation.  This appears both in the UI and in the log:

[image]

`14:2:38 ↓ G:{"name":"iSpindel001","ID":3514868,"angle":48.24492,"temperature":14.9375,"battery":5.338894,"gravity":9.686635}
14:2:38 ↓ G:{"name":"iSpindel001","battery":5.34,"sg":9.687,"angle":48.24,"lu":1519596158,"lpf":0.10,"stpt":1}`

I'm assuming the first line from the iSpindel and the second is what BPL is going to report based on the values in the previous line.

Unchecking the Calibrate by BPL button and saving, results in the exact same data being shown in the UI and the log.  Checking/Unchecking, saving, rebooting, and every combination produces the same result.  It's as if the settings changes are being ignored.  The log entry showing a "Low Pass Filter" of 0.10 also leads me to believe this since I put in a value of .00833.  I'm wondering if 0.10 is a default.

Similarly it seems like remote logging stopped working for me again and just like the above no amount of resaving/changing the settings seems to help.

I'm not sure how to help debug this.  This is using the 2.5.1 thorrax bin

vitotai, 2018-02-26:
The settings is stored in a file named "gdconfig". Please get it by file manager and provide it to help debugging. Thanks.

BTW,

The stability threshold is 1 for 0.0001. Your specification is invalid, and I am not sure what will happen. 
https://github.com/vitotai/BrewPiLess/wiki/iSpindel-Support

rflemming, 2018-02-26:
Alrighty, stability threshold is set to 1.  That doesn't appear to have made a difference.  Here are the contents of gdconfig:

`{"ispindel":true,"cal":false,"tiltw":"","cbpl":true,"gc":"","tc":true,"ctemp":"20","a3":"0.000001051","a2":"-0.000101986","a1":"0.004263415","a0":"0.952348253","lpc":".00833","stpt":"1"}`

vitotai, 2018-02-27:
I do find a potential issue, but I don't have the hardware to verify before end of March. 

in ExternalData.h line 293
`const int BUFFER_SIZE = JSON_OBJECT_SIZE(8);`
change it to
`const int BUFFER_SIZE = JSON_OBJECT_SIZE(16);`

The objects in the JSON files are more than 8.

rflemming, 2018-02-27:
I made that change and built a new bin and that looks like it may have solved it.  Remote logging to ubidots still isn't working (despite working in the past and no changes), so there may still be something else going on, but the gravity sensor config issue seems resolved.

vitotai, 2018-02-28:
Thanks. I've ordered a d1 mini and will be able to test ubidots issue later.

RaceMouse-of-Denmark, 2018-03-05:
That fix did it for me also (on d1_mini).

rflemming, 2018-03-29:
The ubidots things seems entirely random.  I've had it stop working and start working without me changing anything, so I'm not really sure where to go with that one at the moment.  I'm going to close out this issue because the configuration saving issue is resolved.

closed 2018-03-29 by rflemming