vitotai/BrewPiLess #69: Cannot save device configuration

kovalBlazej, 2018-03-01:
I cannot save any device configuration on Wemos D1 mini pro, i just flashed with brewpiless 2.5
It detect temperature sensor, but when i choose function, slot, and click 'apply', and then 'Refresh Device List' slot and function are blank again. It looks like some problem with flash. In logging settings there is "Free Space: -1,073,740,520 Bytes, Used Space: 1,073,740,536 Bytes, Total Space: 16 Bytes "
First i think it is some problem with hardware, but I have same issue on few different boards. On other hand I tried other version and there is still this problem.
I flashed board same way , that previous board that was working fine - using NodeMCU Flasher and binary file from repository.

vitotai, 2018-03-02:
Did you hit “erase setting” button?
https://github.com/vitotai/BrewPiLess/wiki/FAQ

kovalBlazej, 2018-03-02:
If you mean 'Erease EEPROM' button on device config page then yes. Just after that, it is possible to set devices, but they disappear few moments later. 'Backup' result in 'error saving: 500'

Text in logging settings is changing, it is sometimes as above, sometimes "Free Space: 0 Bytes, Used Space: 0 Bytes, Total Space: 0 Bytes " and sometimes "Free Space: 1,073,645,454 Bytes, Used Space: 0 Bytes, Total Space: 1,073,645,454 Bytes "

I also try :8008/format-spiffs but it doesn't help.

kovalBlazej, 2018-03-02:
BTW, I tried it on few boards, but all are from same ali-express seller, so I probably it is hardware problem. Still, I'll be grateful if you help me diagnose what exactly is wrong, maybe I will be able to proof seller, that his chips are faulty.

vitotai, 2018-03-02:
1. The device settings( Pins, sensors) are stored in EEPROM. It is on the flash but not managed by SPIFFS.
2. The SPIFFS should never show 0 bytes used. The very first time it boots, the default network configuration is created and written.
3. I would suggest 
3a. Make sure the power supply can supplies more than 1A. 
3b. Chang a USB cable

jussipu, 2019-09-07:
I have same problem. I can save all other settings but not logging page. Did you find any solution?

vitotai, 2019-09-08:
MQTT is NOT supported in Classic frontend.

jussipu, 2019-09-08:
I'm using 'newui-oled' latest source

vitotai, 2019-09-09:
Try latest 3.6v2(should be r2). MQTT settings is moved to Logging pages. There is a bug in 3.6 when migrating to ArduinoJson V6.

Device setup is stored in EEPROM while other BPL settings are in SPIFFS. However, both are saved on the flash. kovalBlazej's issue seems to be a Flash Layout issue. Wrong flash size of wrong image. I don't know.

kovalBlazej, 2019-09-09:
If I remember correctly, as you adviced, changing the USB cable and reflashing device helped in my case.

bigadz87, 2020-03-05:
I have the same issue.

I have a Rainsun D1 Mini Pro, have flashed multiple version of 3.6 with different usb's, different ports, different machines, and I still cannot get config to stay in memory.

One condition I observe is that I go into device setup, I initialise eeprom, I select devices and move them into slots and functions, refresh list and fine the devices have moved into "installed".

At this point if I hit "backup", I get an error 500, if I then hear a click in my BPL device (sounds like relay firing) and the device reboots itself with no config.

vitotai, 2020-03-05:
Try formatting SPIFFS. Something wrong with your flash, either file system or FLASH.

bigadz87, 2020-03-05:
how do you do this?

bigadz87, 2020-03-05:
sorry, I followed the instructions and connected to brewpilocal / and formatted the spiffs however this did not solve the problem. I still cannot backup config as I get error 500 and then devices reboots.

i have the same rainsun wemos pro as in https://github.com/vitotai/BrewPiLess/issues/101

vitotai, 2020-03-06:
Read this issue from the first comment. All I can help was already stated.