vitotai/BrewPiLess #71: DS18B20 not detected on Sonoff TH10 and BPL 2.5

MikelooM, 2018-03-07:
Hi,

I found that BPL v2.5 (BrewPiLess.v2.5.sonoff.bin, BrewPiLess.v2.5.1.sonoff.bin) cannot detect the original temperature sensor connected to Sonoff TH10. It only shows External Sensor (but no value, Controller PIN - NA). D5 is detected as PIN.
All is fine on BP v2.4.2 - sensor detected on D5, showing correct hardware address and value.
I am running default config and binaries from GH.
Any help/advice would be appreciated.

vitotai, 2018-03-08:
I guess the button feature has something to do with this. I will check it later.

vitotai, 2018-03-12:
I don't find anything suspicious. I tried flashing my D1 mini with Sonoff option, but I didn't see D5 detected.
Maybe you can try latest v2.6?