vitotai/BrewPiLess #72: Cant login

insnhx, 2018-03-14:
Hi, on one of my ferm chambers I had an older version of brewpiless (dont know which exactly) and I changed the display name. After that I couldnt login anymore to the setup or config page. I tried installing the latest version of brewpiless on that board but the display name hasnt changed and the default username/pass dont work. I also tried the eeprom cleaner and reinstalling latest version of brewpiless after cleaning but I get same results (even the display name stays the same)  . Please help :( is there anyway to reset everything to the default values?

vitotai, 2018-03-15:
The old default password is "brewpi" while the newer one is "brewpiless".
If you are not lucky, there are a few ways.
First, connect the USB, build a BPL version with debug option On, and check the log. The password/username will be printed log during power on.
Second, format the SPIFFS file system. There is a esptool.py command to do it. Or, you can find a binary/sketch that formats the file system.

insnhx, 2018-03-15:
Thank you for the reply! I tried the brewpi password but it didnt work. I will try the debug method tomorrow and close the issue. Keep up the good work!!