vitotai/BrewPiLess #73: Are some *.js scripts are missing in htmljs?

geschwaerzler, 2018-03-19:
I'm working on a configuration with 1-wire actors using DS2413. It seems, that device setup has to be augmented for that purpose. When I tried to debug the client code, I discovered some missing files in htmljs:

- js/script-config.js
- js/script-gravity.js
- js/script-logging.js
- js/script-setup.js

In order to build the client code I might augment gruntfile.coffee as well, isn't it?

Best
Gerhard

vitotai, 2018-03-19:
Yes. I missed them. Will upload later.

vitotai, 2018-03-22:
I've uploaded the files. Please check.

geschwaerzler, 2018-03-22:
sorry, files are still missing. Last commit was on 12th March.

vitotai, 2018-03-22:
Sorry, I am in the progress of a new feature. Please check this branch:
https://github.com/vitotai/BrewPiLess/tree/AutoCap/htmljs/src

geschwaerzler, 2018-03-22:
Perfect. And there are other *.js files as well ;-)

I'm going to work with these files and include those that are needed for building the client in a forthcoming pull request.

closed 2018-03-22 by geschwaerzler