vitotai/BrewPiLess #81: Bootloop after few days

Enethion, 2018-04-02:
After few days (2-3) Wemos D1 mini with latest BPL installed fell into bootloop. LED was constantly glowing, device was not available in the network, and neither it was sending any data to remote service. Disconnecting power supply was the only way to break the loop.

This issue might be related to #78

vitotai, 2018-04-03:
I think it has something to do with file system.
I had similar with my controller, currently running. I can't access it now, but it keeps temperature, and the buttons and LCD work fine. Before I lost control over network, I found that BPL failed to update configuration files and logs.
The local logging algorithm needs to be modified/revised for robustness. Before that, I need to check the source code of SPIFFS.

allthepies, 2018-04-03:
I had similar once, BPL was in a bootloop.  I think I raised an issue here about it.  When I plugged in a serial monitor it was complaining about reading from the log file.   I ended up formatting SPIFFS.  
Do you have logging enabled ?

vitotai, 2018-04-09:
I am sorry to hear that.

Although I can't be sure wether it is an issue related to software or not, I would suggest you to get ANOTHER esp8266 AND/OR the power supply.

Enethion, 2018-04-09:
Yeah, I might try that.