vitotai/BrewPiLess #82: Can't connect iSpindel 5.8.5 with BPL2.6 OLED version

Hwerow, 2018-04-06:
Hi Vito,
I am struggling to connect my iSpindel v 5.8.5 with my Brewpiless v 2.6 OLED version. I have set the Brewpiless as at Static IP 192.168.0.150 Gateway 192.168.0.1 and have set up the iSpindel with that address 192.168.0.150.

But it doesn’t show up on the brewpiless.local/ screen!

I have reflashed BPL and iSpindel and it still doesn’t work.

The iSpindel info screen shows the Station IP is reported as 0.0.0.0 but in the configuration it shows as 192.168.0.150 not sure if this is significant or not 
 
Below is the Putty log from the iSpindel – I do not know what the error code 6 means or whether the rest is correct or not

{dl|d|lc|scboodoocplslslpglcg|lbnndd`nod`o{ocl{gbdl`o
FW 5.8.5
2.2.1(cfd48f3)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file
parsed json
offsets not available
parsed config:
{"Name":"iSpindel001","Token":"","Sleep":60,"Server":"192.168.0.150","API":3,"Port":80,"URL":"/gravity","DB":"ispindel","Job":"ispindel","Instance":"000","Vfact":194.56,"TS":0,"SSID":"brewpiless","PSK":"brewpiless","POLY":"-0.00031*tilt^2+0.557*tilt-14.054","aX":0,"aY":0,"aZ":0}
woken from deepsleep, normal mode
Spl 0: 79.26
Spl 1: 79.34
Spl 2: 79.37
Spl 3: 79.38
Spl 4: 79.35
Spl 5: 79.25
Spl 6: 79.29
a: -2556173922282absTilt: 79.33T: 39.30V: 4.12owT: 33.19Gravity: 28.18
After waiting 5103ms, **result 6**
Rescue Wifi credentials
failed to connect
Final-sleep: 60s; RT:7018

I ran Putty for a fair while and it did not connect.

I used Spiceworks IP scanner and the BPL IP of 192.168.0.150 shows up -  so it looks to be available.

To verify that the iSpindel was working I connected to local wifi and Ubidots and it works
Log below
=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2018.04.04 17:01:44 =~=~=~=~=~=~=~=~=~=~=~=
{dl|d|db|ccooloocxl{d{dxgdco|lcggll`ggl`nsncl{ocll`o
FW 5.8.2
2.2.0(f28eaf2)
Worker run!
Boot-Mode: 5
mounting FS... mounted!
reading config file
opened config file
parsed json
applying offsets
parsed config:
{"Name":"iSpindel001","Token":"--------------------------------","Sleep":120,"Server":"","API":0,"Port":80,"URL":"/gravity","Vfact":194.54,"SSID":"SYCONSG","PSK":"----------","POLY":"","aX":-1313,"aY":1225,"aZ":1633}
woken from deepsleep, normal mode
Spl 0: 76.32
Spl 1: 76.36
Spl 2: 76.31
Spl 3: 76.35
Spl 4: 76.34
Spl 5: 76.36
Spl 6: 76.42
a: 440216406-386absTilt: 76.35T: 29.12V: 3.98owT: 26.75Parse error at 1
Gravity: 0.00
After waiting 600ms, result 3
IP: 192.168.0.53
calling Ubidots
{"tilt":76.34843,"temperature":26.75,"battery":3.978617,"gravity":0,"interval":120,"RSSI":-36}Sender: Ubidots posting
POST /api/v1.6/devices/iSpindel001?token=”                            “ HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 94
POST /api/v1.6/devices/iSpindel001?token=”                   “ HTTP/1.1
Host: things.ubidots.com
User-Agent: ESP8266
Connection: close
Content-Type: application/json
Content-Length: 94
Final-sleep: 120s; RT:4695

BPL settings are brewpiless Port 80 and Station
Static IP 192.168.0.150
Gateway 192.168.0.1
Under gravity the iSpindel and calc by BPL are selected 

Suggestions welcomed as to how I can get this working
Peter

universam1, 2018-04-06:
Looks like BPL has an issue to connect against. As you see the generic functionality of ispindel is working, and BPL uses the generic API. 

I would suggest you to open an issue there, and put a link here so others can follow.

vitotai, 2018-04-07:
It seems that the report content of iSpindel grows faster than I had anticipated. The buffer or ArduinoJson is not big enough for the JSON content in your log.

Hwerow, 2018-04-08:
So what would you suggest as compatible versions of iSpindel and BPL that will work together?

vitotai, 2018-04-08:
I have been using the version I built here:
https://github.com/vitotai/BrewPiLess/tree/master/extra

I don't know when Sam added the new fields, and don't know which version will work.
If you can build the code, you can change the buffer size. 
Check ExternalData.h, find processJSON()
`
	bool processJSON(char data[],size_t length, bool authenticated, uint8_t& error)
	{
		const int BUFFER_SIZE = JSON_OBJECT_SIZE(12);
`

Change the "12" to 18, or make 20.

universam1, 2018-04-08:
It is true there are some more data points added, I wasn’t aware that you are accounting for a fixed size of json steam. 

May I suggest you to switch to dynamic json which tends to be suggested for ESP anyway? I think it would make it more resilient for the future. The dynamic json may only fail if not enough ram available which is highly unlikely, but that is simple to catch.