vitotai/BrewPiLess #89: Calibrate iSpindel temperature sensor

ksga, 2018-05-03:
My iSpindel temp sensors seem to be a bit off target - 2,0 degrees to be exact ;)
I've been trying to correct the readings using JSON, but no luck.

The "j" parameter seems to be missing - but that might be by design or? (setting "j" for BPL sensors works fine)

13:58:17 ↓ d:[
{"i":0,"t":1,"c":1,"b":0,"f ":5,"h":2,"d":0,"p":12,"v":22.688,"a":"28FF99C8371604AA","j": 0.000},{"i":1,"t":1,"c":1,"b":1,"f ":9,"h":2,"d":0,"p":12,"v":22.625,"a":"28FF55443816043D","j": 0.000},
{"i":2,"t":3,"c":1,"b":0,"f ":2,"h":1,"d":0,"p":13,"v":0,"x":1},
{"i":3,"t":3,"c":1,"b":0,"f ":3,"h":1,"d":0,"p":14,"v":0,"x":1},
{"i":5,"t":1,"c":1,"b":0,"f ":6,"h":5,"d":0,"p":0,"v": 24.688}]

vitotai, 2018-05-04:
I thought iSpindel has the calibration setting, or maybe not.

ksga, 2018-05-04:
It might. I just haven't been able to find it ☹️
Let me check tomorrow with a fresh wipe/install...

universam1, 2018-05-07:
@ksga You might be blaming on the wrong sensor, consider that. Actually, the DS18b20 is calibrated from the manufacturer up to an error of +/-0.5K, and I personally run a few verifications and can confirm this was always true. 

In general, the so called 'reference' people are using are off rather than that sensor! Hence it is better to actually question the other side if they are off, we never had a situation where the DS was actually to blame

ksga, 2018-05-07:
@universam1 True.
I used 2 DS18's connected to BPL (as seen in the screen above). They seem to correspond quite well.
Also I found an old Labortherm mercury thermometer. Of course it's hard to read with 3 digits - but it agrees with the sensors connected to BPL.
The readings on the iSpindel information page are the same as those recorded in BPL.

Of course it could be faulty DS18's - but it seems odd that I have two with the same error margin.

universam1, 2018-05-07:
@ksga what interval are you running? Consider if lower interval than 60s will lead to self heating because of the ESP. Maybe that is your issue?

ksga, 2018-05-08:
Normally I run it at 600 seconds. On the test I just did the sensors are taped together, so readings should be fairly similar.

Maybe I have a couple of faulty sensors - but an option to calibrate would still be nice. Easier than un-/resoldering/recalibrating the spindel. 

Of course a faulty sensor might have a non-linear error margin, but with the used temperature range I haven't been able to notice it (15-30 degrees tested).

ksga, 2018-05-08:
Yipiii - thanks a lot @vitotai 😁

vitotai, 2018-05-09:
Allow me to remind you that the implementation is in the source now but not in the binary.
Test of the latest modification is on-going.

dsgfh, 2021-04-06:
I know this is an old thread, but I've just found my ispindel also out by+2C against other calibrated probes. Did anyone solve this issue?

indypipps, 2021-04-29:
UP: me too... :(