vitotai/BrewPiLess #92: Wemos Wroom-02 file system issue

thatinfidel, 2018-05-25:
Not sure if this is a true issue or just something strange with the device I am using.
Flashed the latest BrewPiLess.v3rc4.oled firmware. Trying to do some local logging, and I get very strange readings regarding free space. I am unable to actually start local logging, I just get Error 404 every time.
I believe the device only has 2megs of storage total, so that could be the issue, but it still seems very strange how it is behaving.

The snippet below is simply based on refreshing the page and capturing the output.

Free Space: 1,074,796,709 Bytes, Used Space: 48 Bytes, Total Space: 1,074,796,757 Bytes

Free Space: 1,073,645,477 Bytes, Used Space: 0 Bytes, Total Space: 1,073,645,477 Bytes

Free Space: -1,070,652 Bytes, Used Space: 1,074,796,724 Bytes, Total Space: 1,073,726,072 Bytes

Free Space: 1,073,659,104 Bytes, Used Space: 0 Bytes, Total Space: 1,073,659,104 Bytes

quimnut, 2018-05-25:
Have you tried formatting it? 
http://brewpiless.local:8008/format-spiffs

thatinfidel, 2018-05-25:
yes that was the first thing i tried. same result. Also it won't save anything else either. I can get it to save my wifi credentials from config mode, but once it's running it loses all settings on next boot. If i try saving any settings that require a restart it will just reboot into config mode, so i have to connect to the brewpiless ssid and re-do all settings in order to make it work.

vitotai, 2018-05-25:
The default configuration assumes 4MB. I didn't trace the source of SPIFFS and am not sure what would happen. 

I would suggest to build the code with right configuration.

thatinfidel, 2018-05-25:
that seemed to do it. Thanks for the help!

closed 2018-05-25 by thatinfidel