vitotai/BrewPiLess #93: Can't get ispindel to work as beer temp sensor.

tblok, 2018-05-25:
It seems as I can't get the ispindel to work as the beer sensor for temperature.

I am getting the values as aux temp ok, also tilt and battery shows ok so there is nothing wrong with my ispindel.
Also in the device setup i set the external sensor to beer temp, but it never gets any value but 0.

I have tried the  default image for 3.0rc4 and also 2.4, 2.5(final) all with the same result,

tblok, 2018-05-28:
Cool :) I will try this after i finish my latest brew on 2.5

solbero, 2018-07-31:
It doesn't seem like this issue is fixed in v3.0. All the data from the iSpindel displays correctly in Brewpiless, except for the temp readings. The readings are displayed correctly on the iSpindel info page, but it seems like they are not read by Brewpiless.

kynsi, 2021-01-11:
This is still an issue. Having the same problems with the latest release (4.1).