vitotai/BrewPiLess #94: Remote access, outside my local network

ferchinas, 2018-05-31:
Hello, thank you very much for the wonderful job.
I have BrewPiLess in my Wemos D1 mini running, but I want to be able to access from outside my local network.
Probe opening and redirecting ports 80 and 81 from my router.
When trying to access from the navigator with my external IP, of the form xxx.xxx.xxx.xxx:80 or 81, I recieve connection error.
Port 81 ERR_CONNECTION_REFUSED
Port 80 ERR_CONNECTION_TIMED_OUT

Could anyone help me?
Thank you very much

vitotai, 2018-06-01:
If it works fine when your phone/computer is INSIDE local network, then your router might have something to do with it.

ferchinas, 2018-06-01:
Hi Vito:
Thank you very much for your answer.
Investigating a bit my router I found that port 8080 uses it for its configuration.
I also found that port 80 is used by the WIMAX device of my internet service provider.

So I think the only option would be to change the port of BrewPiless.
Can I do it from configuration, or do I need to make a compilation?
Thanks a lot,

vitotai, 2018-06-02:
You can, but I think the DMZ setting usually allows port mapping. Anyway, you can change it in Setup page.
https://github.com/vitotai/BrewPiLess/wiki/System-Setup

ferchinas, 2018-06-03:
Hello vito: Thank you very much for your response, I have tried with DMZ but without success. My Internet connection is by air, the active antenna that serves with internet to my router has a configuration page on port 80.
I managed to make it work by accessing a different external port "8088" and redirecting it to internal port 80.
Thank you very much

closed 2018-06-03 by ferchinas

Paiakan, 2018-06-14:
Hey, @ferchinas would you mind to share exactly how you did it? I have the same issue and it seems the only available port is 1723 but until now I had no success. I already tried to change the port on BrewPiLess setup to 1723 but it still doesn't work. 

Any help will be appreciated.

ferchinas, 2018-06-14:
Hi, @Paiakan 
The first thing I did was, I assign a static IP to my brewpiless. In my case 192.168.1.104.

[image]

Then I redirected the ports:
8008 (external) -> 8008 (internal)
8088 (external) -> 80 (internal) because the external port 80 was already used in my case

[image]

To access from the internet, you have to type in your navigation bar: xxx.xxx.xxx.xxx:8088 where xxx.xxx.xxx.xxx is[ your external ip](https://www.whatismyip.com/my-ip-information/)

I hope it serves you, regards

Paiakan, 2018-06-14:
Thanks for your help, hermano. Still have some doubts:
1 - The forwarding service from 8008 (ext) to 8008 (int) looks disabled (inhabilitado). If it's disabled is it really necessary?
2 - What did you use at the Brewpiless setup config port, 8088?

Thanks again!

ferchinas, 2018-06-14:
1-Sorry, port 8008 should be enabled, I disable it to do a test. Brewpiless defines it in the file "Config.h" line 449. I think it's only necessary for updates. To access the pages "/ filemanager" and "/ systemupdate".
2-I chose port 8088 at random, because in my case port 80 is already used. you can use the one you want that does not conflict with another service in your network. For example, port 1723.