vitotai/BrewPiLess #95: Multi languages support

ferchinas, 2018-06-01:
Hello:
I would like to know if it is on the wish list to multi language support.
I am willing to collaborate in Spanish translation.

LucLN, 2018-06-01:
Hi,

if so, I would contribute in the Luxembourgish translation .. 

(although I'd probably be the only one using it :-)

LucLN, 2018-06-01:
ok .. if it motivates, I'll do German and French too ..

vitotai, 2018-06-02:
Well. I've tried surveying the internationalization of html/javascript, and unfortunately I haven't found a good way. I don't think dynamic processing HTML on ESP8266 is a good idea.
Thanks to Tom, now that the HTML/Javascript files are processed by grunt, internationalization might be done by grunt. I just have to find a grunt plug-in to do it. Unfortunately, I am not familiar with grunt, so it might take some time.

ferchinas, 2018-06-05:
Hi Vito:
I also agree that overloading ESP8266 with dynamic pages is not a good idea.
Never use grunt, but it seems that it is a very good tool to generate pages in different languages. I found [this template](https://github.com/BBCVisualJournalism/grunt-multi-lang-site-generator) that can be useful.
The idea is that later, whoever wants to use a language other than English can download the translated pages to the file system?
regards

vitotai, 2018-09-13:
I've tested with multi_lang_site_generator.
It seems to work. However, I am not familiar with grunt and couldn't make the build automatically.
If someone can provide translation files, I will merge and build the specific language web front pages. 
Please check the locale files here:
https://github.com/vitotai/BrewPiLess/tree/Localization/htmljs/src/locales

ferchinas, 2018-09-18:
Hi Vito:
Very good that you have been able to add internationalization! My brewing partner will be very happy with this.
I started making the translation into Spanish, I found minor details in the file "english.json" and made a pull request (first time I do it, I apologize if I made an error).

I'm not familiar with the "Beer profile" options. Which condition evaluates the "Either" option because that word has several translation possibilities.

"dlg_poly_cal": "Cal'ed" = "Calibrated"?

When you generate the Spanish front end I will find errors and problems with the size of the strings and I will have to correct them. ;)

I hope it is useful,

Thank you very much
Fernando

vitotai, 2018-09-20:
It is "calculated". 
I just updated and built Spanish version. Please check it. I don't read Spanish. Gracias.

ferchinas, 2018-09-21:
Hi Vito:
It looks great in Spanish!
Only I will have to make some corrections.

What do you think if we also process the javascript files to translate the alert () messages and dynamic changes in the text of the pages?

ferchinas, 2018-09-26:
Hi Vito:
Yesterday I was seeing how to execute grunt and also locate the alert () and confirm () messages.
I did not pullrequest yesterday because I did not get to try it.

Unfortunately it seems that we did the same job.
Today I will also do the pullrequest on the translation and you decide if it is preferable to use your version or mine.

There are some texts that apparently come from the controller, do you think we can use the same tool to translate them?
[image]

vitotai, 2018-09-27:
Yes. they can be done.
The state at right corner is defined in Javascript. 

The Mode and status are extracted from LCD text. They are parsed and ready to be localized. 

The "status line" at top left corner needs some care. 
All possible strings are:
            Idling for 00h00m59
            Temp. control OFF
             Door Open
             Heating for 00h00m59
             Cooling for 00h00m59
             Wai  to Cool 00h00m59 
             Wait  to Heat  00h00m59
             Waiting for Peak
             Cool Time left 00h00m59
             Heat Time left 00h00m59
             Unknown status!
A printf like syntax works best. However, the "hour" part is not mandatory, which means pattern like %02dh%02dm%02d is not the best solution. A easy solution is to define two format, long and short, like %02dh%02dm%02d and %02dm%02d. How do you think?

ferchinas, 2018-10-02:
Hi Vito:
I am using the translated version (especially in the Tom's front end) and it looks great.
Very good work,
Thank you very much

closed 2018-10-02 by ferchinas

vitotai, 2018-10-06:
hi ferchinas,

I added some terms for network selection. Please help to review Spanish translations. They are all in config.htm. 
Long story follows.
I tried to make Tom's frontend as default. One problem is that the file sizes are bigger and the whole image will be too large. To support OTA update, the program show be less than half of the program space, that is around 500k in current default settings. However, using Tom's frontend, the image will over the limit. Therefore, extending the program space is necessary. Fortunately, framework 1.8.0 already has the required layout, although you can manually add it in previous version. However, I had one small issue of WiFIManager in framework 1.8.0 and was forced to ditch WiFiManager, which I had planned to do in next version.

ferchinas, 2018-10-10:
Hi Vito:
how are you? It seems that there are big changes for the next version!
Definitely the Tom's front end is more attractive and intuitive, i think there is a good idea to make it default.
I found some details in the Tom's front end, I would like to improve it, for example the style of some buttons does not correspond with the general style. What do you think if, when I have a little time, I make some changes and propose them?